Mr-MIBonk/M.I.B._More-Incredible-Bash #316: No possibility to change USB configuration coding

hadez16, 2023-08-18:
There exist cars with MIB2 High equipped but without having an Apple-capable USB port mounted.

In such cases it can be (or has to be) necessary to change to coding of the USB from "iPod" to "FULL" in order to make Android Auto work.

In the current version the USB configuration is only a display (read-only) and not changable.

[image]

It would be a huge improvent to be able to **change** that coding!

zzz08900, 2023-08-19:
Is it really possible to modify long coding through MIB headunit? 

From what I gathered you'll need VCDS/ODIS to modify long coding.

hadez16, 2023-08-20:
The Toolbox already changes the coding today, when you apply the aio patch.
But it always sets the USB configuration to "iPod".