Mr-MIBonk/M.I.B._More-Incredible-Bash #318: GEM very small and unreadable

werdck, 2023-08-23:
Disclaimer: This might be unrelated to M.I.B.

After installing M.I.B., my GEM screen is very small, like it was zoomed out to ~30% of display size.
Is this a known issue, or at least something that can be changed / fixed?

Log and photo attached.

[MU1326-JTH-07607.12.183409002C.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/12420608/MU1326-JTH-07607.12.183409002C.log)

[image]
[image]

werdck, 2023-08-27:
Resolved.
My MHI2 unit was incorrectly coded (Bytes 01 and 02). After fixing that yesterday, menu is "fullscreen" again.

Closing.

closed 2023-08-27 by werdck

h4n1793, 2025-04-30:
Hey sorry that this thread has been closed but how did you fix this issue I have updated my firmware and now my screen is the same size as what’s in your picture

werdck, 2025-05-01:
For me it was stg 5F, bytes 01 and 02 that were misconfigured. Those hold your car's model and engine type for display in the MMI. You should search for coding strings of your vehicle model, module 5F Infotainment.

h4n1793, 2025-05-01:
What would I need to change them to I have a Audi a3 2017 model with nav plus

dabrase7, 2025-06-08:
Hi, after installing MIB and updating my Audi A3 MMI, the GEM is small and unreadable as well. I don't really know what stg 5F means. What values are correct for bytes 01 and 02? How do you search and configure that? Is it using VCDS or ODBII? Please, I need to solve this issue to continue maps installation and other features using GEM, thanks

h4n1793, 2025-06-08:
On the odb11 search the 5f module it’s the multimedia module select long
coding and under bytes 01 and 02 you need to recode it to your vehicle
class and derivate if you know them what car do you have