Mr-MIBonk/M.I.B._More-Incredible-Bash #339: Issues after conversion on MIB1 High

chefranov, 2023-09-28:
I converted **MHIG_US_AU_S2016 MU0403** to **MHIG_EU_AU_K1555 MU0415** according this manual https://www.drive2.ru/b/652054101932133902/ 
**And now I have three errors:**
B201200 - MMI languages
B200087 - Software incompatible
B126CF1 - Navigation db not available

I have two languages in the menu: English and Spain. Also I can't install maps.
**Installed (legal) FECs:**
Navigation NAR (0211001E)
Navigation EU (021000EE).

Car: Audi A3 8V 2016 e-tron (without B&O)
Unit: 8V0035038C

How to fix it?

chefranov, 2023-10-18:
Fixed myself.
B201200 - installed eu langs dataset with ODIS. By some reason menu in M.I.B. is not appear where I can install it
B200087 - reinstall firmware with selected all items (except of screen engine)
B126CF1 - cleared DB and installed EU maps

closed 2023-10-18 by chefranov

aledepaoligampel, 2023-11-17:
Hi @chefranov! you are in EU or NAR? I have the same problem, I cant install maps from RoW (im in Argentina).