Mr-MIBonk/M.I.B._More-Incredible-Bash #341: TMC for UK

BooBoo124, 2023-09-30:
Hi,
I've run this version and gone on to the TMC page hoping to reactivate TMC on maps here in the UK. Thinking I was going to be changing the value from 15 to 14 as in the chart I've seen. However all I see are patches for Russia China and France. 
Do I need VCDS to do this?

BooBoo124, 2023-10-04:
No help or support?

catatonicChimp, 2023-10-16:
Can't fix without help... 
5min google says:
https://www.ttforum.co.uk/threads/fixed-rds-tmc-no-longer-working-in-uk.2032197/post-9629703
If you have a value below 32768, change to 14.
If you have a value above 32768, change to 32782.

BooBoo124, 2023-10-17:
I saw that. Was hoping it could be done a different way apart from vcds. My cable has been locked!

catatonicChimp, 2023-10-17:
In M.I.B, no, but looks like you can do it with MIB2Toolbox... though last time I touched that it broke a bunch of my default settings and had to recode my unit. at which point you probably still need a vcds or obdeleven etc.. 
https://github.com/jilleb/mib2-toolbox/

https://github.com/jilleb/mib2-toolbox/blob/e689d120d072df7db71f11463e9bd5a272d6ef61/Toolbox/GEM/mqb-rccAdaptation.esd#L4

BooBoo124, 2023-10-25:
Can both mib toolbox and M.I.B be  installed on the unit?

BooBoo124, 2023-11-02:
I've bought an Obdeleven and code it

closed 2023-11-02 by BooBoo124