Mr-MIBonk/M.I.B._More-Incredible-Bash #354: Help to install new maps (may have to delete FEC 02300024 FEC 023000ee also available)

specialex, 2023-10-28:
Hi there,
I already patched my system and FEC 02300024 is still visible but FEC 023000ee also visible.
Got a problem to install new maps may because of I have to delete FEC 02300024 but I don't know how to.
Many thanks for your help.

Have a look to my screen shots.

[image]
[image]
[image]
[image]
[image]
[image]

[MU1329-JTB-07414.09.171171027C.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/13196114/MU1329-JTB-07414.09.171171027C.log)

Kind regards and many thanks in advance

Alex

ccrespo72, 2023-11-09:
Hola,donde te descargarte la actualizacion??
tengo el mismo problema y no puedo actualizar mapas.
Gracias

Soldafaya, 2024-02-13:
Hello,
Have you found a solution ? I have the same problem 
Thanks

specialex, 2024-02-14:
At the end this is was not a problem, the problem was I tried to use a Mac to prepair the SD card for the update. This doesn’t work you really need to use a windows computer. 
hope this helps a bit.

Soldafaya, 2024-02-14:
Ok I'm in the same situation I use mac to prepare the SD. 
Thank you

Soldafaya, 2024-02-27:
bonjour 
j'ai fait avec un pc mais la nouvelle carte /map n'est toujours pas reconnu lors de la demande de mise à jour 
avez vous une idée ? Merci