Mr-MIBonk/M.I.B._More-Incredible-Bash #356: MHI2Q maps not installing

woodyash, 2023-11-02:
Good Afternoon 

I'm trying to carry out an EU conversion on a JP MHI2Q unit however having issues getting the new maps to install and wondered if you could help. 
I've used the tool and carried out the conversion, installed new firmware and patched the unit to MHI2Q_ER_AUG22_P5092 MU SOFTWARE 1329 with all required FECs legal. My issue is every time I try to install new maps in REM it keeps saying Speech check timeout and if I update it through MMI settings it just keeps saying searching with no option to update.

Thanks 

Ash

traycho, 2024-03-31:
Same issue, did you solve it ?

woodyash, 2024-03-31:
Reinstalled UK firmware then in MIB tool do a clean up (can't remember what its called in advanced). Then retry. May need to resection your harddrive via diagnostics to give enough space as well.

traycho, 2024-04-01:
Perhaps you are talking about `Advanced Settings -> SWDL CLEANER` ?