Mr-MIBonk/M.I.B._More-Incredible-Bash #357: Gracenote v23

BooBoo124, 2023-11-02:
Hi I'm running a patched version of MHI2Q_ER_AUG22_P4212
I had gracenote v21 installed. I updated to V23 now I have no album art via Bluetooth stream. Is there away to downgrade back? It won't allow me to via the usb I stall method or do I need to update my firmware?

Schmisoo, 2023-11-02:
> Hi I'm running a patched version of MHI2Q_ER_AUG22_P4212
> 
> I had gracenote v21 installed. I updated to V23 now I have no album art via Bluetooth stream. Is there away to downgrade back? It won't allow me to via the usb I stall method or do I need to update my firmware?

Had the same issue after i updated from v17 to v23. Just installed v17 back after I could not find any solution in the web...

BooBoo124, 2023-11-02:
How did you roll it back to previous version?

Schmisoo, 2023-11-02:
> How did you roll it back to previous version?
> 
> I tried to go to V22 but it was greyed out and wouldn't let me

I just downloaded die v17 from MIBSolutions and installed it as a update. Maybe you need to clear the cache of sw updates with m.i.b? But that is just a guess

Schmisoo, 2023-11-02:
I installed it with the software update function from "Systempflege" (german). But I actually don't know if that makes a difference. I think we need a advice from an expert. I also asked multiple times in the Telegram chat for help but didnt get a single answer 😀

BooBoo124, 2023-11-02:
Do you have a link to that software  please.

Schmisoo, 2023-11-02:
https://mibsolution.one/#/1/9/MHI2%20-%20HARMAN/Gracenote

V17

BooBoo124, 2023-11-02:
I ment the Systempflege

Schmisoo, 2023-11-02:
That is the native software update function in the user settings of the system.

BooBoo124, 2023-11-02:
Oh got you via the mmi set up tab

Schmisoo, 2023-11-02:
> Oh got you via the mmi set up tab

Did that work?

BooBoo124, 2023-11-02:
I will try tomorrow very bad weather here at the moment

Schmisoo, 2023-11-02:
Informed other users

https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/discussions/211

BooBoo124, 2023-11-03:
V21 went on ok via the system maintenance update menu and worked fine.  I  then put V23 back on via the maintenance menu. Nothing all gone again

Moody06, 2024-01-13:
Hello I tried to install on S5 B9  the Gracenote2 V21 because I couldn't find the V22 but the metadata file miss on the V21...
Someone can help with a link of Gracenote2  V22 or a metadata file for V21 please?

BooBoo124, 2024-01-13:
[https://mega.nz/file/JpA2zbJT#U-NlDCLP4fGCqW7GB9Yv3ztqonpEc7GhmuQ2AE9Rtts](url)

V22

Moody06, 2024-01-13:
> [https://mega.nz/file/JpA2zbJT#U-NlDCLP4fGCqW7GB9Yv3ztqonpEc7GhmuQ2AE9Rtts](url)
> 
> V22

Thanks for it! 
The V23 issue is still relevant?

BooBoo124, 2024-01-13:
it is for me

Moody06, 2024-01-13:
> it is for me

Do you mean it is not work correctly on your car or you can’t install the V23?
What is your car?

Moody06, 2024-01-13:
> > it is for me
> 
> Do you mean it is not work correctly on your car or you can’t install the V23? What is your car?

I found my answer on your previous post sorry:

« Hi I'm running a patched version of MHI2Q_ER_AUG22_P4212
I had gracenote v21 installed. I updated to V23 now I have no album art via Bluetooth stream. Is there away to downgrade back? It won't allow me to via the usb I stall method or do I need to update my firmware? »

Moody06, 2024-01-13:
> it is for me

Did you try to update your MMI Firmware? b
Because your version is probably too old for Gracenote2 V23?…
My version is MHI2Q_ER_AUG22_P5092_MU1329 on 5F module
I will give I try to the V23 and see if it work on my car

BooBoo124, 2024-01-13:
I didn't update the firmware.

Moody06, 2024-01-13:
> I didn't update the firmware.

After reading this post with an updated Firmware:

Had the same issue after i updated from v17 to v23. Just installed v17 back after I could not find any solution in the web...
Audi A3 MHI2_ER_AU37x_P5089_MU1326 Patched

 I will NOT try to install the V23. Wait till V24 is out…
Thanks for your help 🙏

BooBoo124, 2024-02-07:
[https://www.drive2.ru/l/667391739383802682/#comments](url)

Gracenote 24

ilopezv, 2024-03-27:
I confirm that V24 works well on MHI2_ER_AUG22_K3346_MU1438

Issoudotpng, 2024-04-17:
> I confirm that V24 works well on MHI2_ER_AUG22_K3346_MU1438

Do you still have your file and if so could you please upload it to me somewhere ? because from mibsolutions it keeps downloading a corrupted copy of the archive

BooBoo124, 2024-04-17:
[https://mega.nz/file/I85jiTaA#_O9j1BLcGZMqdeMefJ_P2JsvR9vdwPPx11-xZWTWaTw](url)

Issoudotpng, 2024-04-17:
> [https://mega.nz/file/I85jiTaA#_O9j1BLcGZMqdeMefJ_P2JsvR9vdwPPx11-xZWTWaTw](url)

Thanks a lot my guy 🤗

BooBoo124, 2024-04-17:
Search drive2.ru gracenote 24