Mr-MIBonk/M.I.B._More-Incredible-Bash #361: Not an Issue but a question...

spaceprograms, 2023-11-23:
Hi Mr-MIBonk,

First let me say thanks for creating this software and for making it freely available!

Sorry but I couldn't find an email address to contact you so that's why I'm asking my question here.

Is it OK to run any of these patches more than once?  The reason I ask is because I'm not sure that I allowed the CarPlay activation one to run to completion? CarPlay does work, I don't have the ASI button in the menu (but I think my car is too old for that). However, sometimes my phone doesn't connect to CarPlay when I plug it in to the USB. So I thought perhaps if I run the patch again it may fix that? I didn't want to try it, in case I can't overwrite something that's already there? Also, if I can run it again, would I need to run the SMV Fix again after? Thanks in advance.

Schmisoo, 2023-11-23:
Mostly a cheap or bad hsd cable is the reason for connectivity issues. Did you upgraded the data USB port by ur self or was it already build in your car?

spaceprograms, 2023-11-23:
Hi not really answering my question but thanks anyway. The USB ports are already in the car. It’s a 2015 TT.