Mr-MIBonk/M.I.B._More-Incredible-Bash #362: MHI2Q_ER Installation

zverko76, 2023-11-24:
Hi, is it possible to install  on MHI2Q_ER_AUG22_P5152?
I saw on YouTube that someone installed on the MHI2Q_ER system. I tried the same but it didn't work.
It came out on the display to go to diagnostics and do Start version compare. I solved that through obd11 with the B201A calculator.
I rebooted the system but the MIB didn't install. What did I do wrong?
[image]

Schmisoo, 2023-11-24:
Read the installation instructions of m.i.b. before doing anything

zverko76, 2023-11-25:
> Read the installation instructions of m.i.b. before doing anything

I did everything as it says in the instructions and it didn't work.
[image]

closed 2023-11-26 by zverko76

DipeshChunilal, 2024-02-14:
i am struggling to find this:MHI2Q_ER_AUG22_P5152 software to download FROM MIBSOLUTION as currently running MHI2Q_ER_AUG22_P5092.
can someone help?

koziolk, 2024-04-05:
@DipeshChunilal Was MHI2Q_ER_AUG22_P5152 even released? I wasn't able to find this version but I saw some cars had it installed.

neoalius, 2025-05-23:
following ..

and If some can send me the down load link for MHI2Q_ER_AUG22_P5092. It will be greatly appreciated