Mr-MIBonk/M.I.B._More-Incredible-Bash #365: Activate smartlink SWING + (octavia3)

GhostSVK, 2023-11-26:
hello, please, is it possible to activate smartlink on the SWING+ radio?  the radio does not have navigation.  I am attaching photos.  please, if anyone has a procedure that will work, direct me.  thank you very much for every response.
[image]
[image]
[image]

igorstefaniak, 2024-04-06:
i have the same problem with smartlink, did you solved it?

lgdvp, 2024-07-15:
Anyone can do this activation? Please….

Godovic, 2024-10-20:
Still nothing for this version?

Djpd66, 2024-10-26:
i to waiting :(