Mr-MIBonk/M.I.B._More-Incredible-Bash #367: Not possible to convert to EU MIHG_US_AU_S2014 neither K2019

kubaawwcz, 2023-11-28:
Not possible to convert to EU MIHG_US_AU_S2014 using M.I.B script 

I have updated SW from S2014 to K2019 and tried again, but I have received the same error

M.I.B LOG file and screenshots attached 

Backup updated on M.I.B solutions here https://mibsolution.one/index.php/f/196876

I had also problem with M.I.B installer for this MU (using latest vrsion 3.6.0), GEM was not activated, needed to use for that OBD11 

[image]
[MU0394-JTB-07427.04.1585900158.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/13493411/MU0394-JTB-07427.04.1585900158.log)

[image]

[image]

kubaawwcz, 2023-11-30:
Issue solved
1. I have used newest verion of MIB not relased yet 3.7.1
2. In 3.7.1 GEM Launcher works, same as EU coding for MIHG
3. I have installed EU firmware in User defined mode but with out changing any of proposed setup (only 4 positions out of 13 possible)
4. EU firmware installed but EU languages were not visible after reebot
5. I have instaled firmware again now selecting all 12 out of 13 possible possitions (without FC1H37xE)
6. EU languages were not visible again 
7. After that I have decided to used VCP and load EU languages datasets
EU languages are visible now

kubaawwcz, 2023-12-04:
A3_MIB_8v0035036.zdc

bahomoncik, 2024-09-29:
Hello, I’m completely new here and trying to do the same thing, I understand that the steps that i need to do are the following, correct me if I’m wrong.
1) install the latest US software , in this case MIHG_US_AU_S2019 (can I do it normally using the red menu ?)
2) using vcds activate the GEM 
3) install M.I.B 
4) use the eu_conversion function in the M.I.B (will it automatically download the eu software version or I need to have it already downloaded on the sd ?) 
can’t I just install the EU version from the red menu after updating the latest US version by turning off the regions lock ?