Mr-MIBonk/M.I.B._More-Incredible-Bash #368: VWG13 VW MIB2.5 Image MOD - Android Auto Icon is not replaced when plugged in via USB

JannF, 2023-12-03:
**Describe the bug**
Upon connecting an android phone with android auto, the skinned icon tile of App-Connect gets replaced by an unskinned (stock) Icon of an Android Auto Tile.
See attached screenshots.

**Check known issues before reporting**
none found, so I seem to be the first to report this

**Add M.I.B LOG file to your report**
[MU1367-JTH-07614.06.18340401F5.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/13540063/MU1367-JTH-07614.06.18340401F5.log)

**Screenshots**
[image]
[image]

**Additional context**
Unit is a retrofit.

Also, removing the MOD seems to not work. See attached log.
The two skin directories do not get deleted as per ImageMod script, but it seems to hit the first case instead and tries to find the Image backup as for Skoda Units, doesn't find it and logs it.

JannF, 2023-12-03:
doh... uploaded the wrong screenshot in the second image.

Here is the right one.

[image]

JannF, 2023-12-11:
The same happens with carplay.

kreshnikislami, 2024-09-30:
can u help me doing this theme i do get a error! i have converted g11 to g13 now i have 9.2 inch display MHI2_ER_VWG13_K5425, but when i touch install the mod i have a error! can u help with that? i really wanna do it that mod

marcel151, 2024-11-18:
Mod works fine with MHI2_ER_VWG13_K4525 in my case. But I also have the problem with missing Car Play Icon.