Mr-MIBonk/M.I.B._More-Incredible-Bash #370: Header size on unit - A0A3EE00 - does not match patch files size

LC22UBI, 2023-12-09:
Hi.

In first place, I have to thank you all for your work!

Now I can tell you my problem:
My radio workes fine after VW service do something to my FECs file (maybe some update).
I have all things unlocked: MAPS, VOICE, Android Auto, PHONE, Bluetoth, DVD... everithing! But after a service in VW official garage the maps don't load (invalid data) The phone dont work, the bluetooth cannot turns on... only  DVD works fine and normal radio. 
I verify the FECs and saw that all are "Temporary Removed"
Then I downloaded last version of M.I.B., downloaded the last patch available to my FW (MHI2_ER_VWG11_K3342_MU1427_PATCH) and  try to run again the patch, but I receved this error:

...
MMX-flash-done!

OK: Patch folder /net/mmx/fs/sda0/patches/MHI2_ER_VWG11_K3342_MU1427_PATCH found on SD card

File MU1427-ifs-root-part2-0x00ba0000-98A3EE00.ifs found in patch folder
File MU1427-ifs-root-part2-0x00ba0000-A0A3EE00.ifs nedded to patch this unit
Header size on unit - A0A3EE00 - does not match patch files size - A4A3EE00 - or stock - 98A3EE00 - image

....
And still

Now I want help to solve my problem, please!...
This process created new backup files that i will try to attach here later. 

Thanks a lot.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Best regards,

Luís Correia

guizmo84170, 2024-05-26:
hi do you have find a solution ?

LC22UBI, 2024-05-29:
Yes.
I went to vw to solve my problem, but the technicians of my garage aren't ready to this kind of software problems. So i ask permission to copy original sd card they had and make some contacts. My VW garage assured the payment of the problem they caused. 
I got a new friend in Lisbon that try to solve the problem but only solve half part. He told me that my vw garage unlock the radio with my VIN, but clean the info file, somehow... maybe because they try to dowload software to my old radio version. So i tell him that i had original SW from VW and then he makes some changes in this original SW SD card and reinstall all. After that, everything works again and my maps still there with the last version. I think the installation fix every damaged file, but the keep the maps versions.

Problem solved.