Mr-MIBonk/M.I.B._More-Incredible-Bash #371: Question:

xirixiz, 2023-12-17:
**Describe the bug**
I recently made a switch from BMW to Porsche. While I got it all clear for BMW, this is rather new for me to explore for Porsche, which is fun! This project is awesome but I have one concern about the SW version I`m having currently: MHI2_ER_POG24_K5137_1417 (see sreenshot). In the compatibility matrix I cannot identify this version, does it mean I cannot use M.I.B., or can I still use it?

**Check known issues before reporting**
Make sure you have checked here:
https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/wiki/Prepare-M.I.B-SD-card-&-install-M.I.B#%EF%B8%8F-trouble-shooting-%EF%B8%8F

**Add M.I.B LOG file to your report**
N/A

**Screenshots**
[image]

**Additional context**
N/A

xirixiz, 2023-12-17:
I just noticed ```MHI2_ER_POG24_P5137_1417``` in the list, but as far as I can find, this release doesn't exist. I believe this should be the release I`m currently having. ```MHI2_ER_POG24_K5137_1417``` (So K instead of P). Would be nice if this can be confirmed and adjusted accordingly. 

I would also love to create a PR for this if what I`m stating is correct.

xirixiz, 2023-12-17:
I just gave it a try and it worked like a charm! I believe the overview of the releases contains a typo as stated in my previous comment, so leaving this ticket open so it can be fixed/updated.

[image]

vinnyvinegar, 2023-12-18:
I'm having a similar issue as you. My version is K3300 but there's only P3300 in the patches list. Did you simply rename the specific patch folder to match your version or did the install complete without any changes?

xirixiz, 2023-12-18:
Hi @vinnyvinegar, unfortunatly, it's slightly different to your situation.

For me my sys ID is ```MHI2_ER_POG24_K5137_1417```, but in the compatibilitylist it showed ```MHI2_ER_POG24_P5137_1417```. But looking at the ```20230926_M.I.B_Patches```, the folders was properly named with a ```K```. 

Looking at ```20230926_M.I.B_Patches``` I only see the patch for ```P3300```, so this is the difference. Do you understand what I`m trying to say? If not, I can elaborate a bit further. I guess someone with a bit more expertise can tell something usefull about it. I would not rename the patch folder. My guess is that it potentially could screw up your system.

xirixiz, 2023-12-18:
@vinnyvinegar Can you provide your full software version, like I have shown in the screenshot?

vinnyvinegar, 2023-12-18:
> @vinnyvinegar Can you provide your full software version, like I have shown in the screenshot?

I understand.Sure, it's 

> Hi @vinnyvinegar, unfortunatly, it's slightly different to your situation.
>
> For me my sys ID is `MHI2_ER_POG24_K5137_1417`, but in the compatibilitylist it showed `MHI2_ER_POG24_P5137_1417`. But [...]

I understand. My full version is MHI2_US_POG11_K3300 MU0809

I tried doing an update in the car using a hotspot but an update wasn't found, in the event my version is too old.

vinnyvinegar, 2023-12-18:
@xirixiz I was able to install the AIO based off of these instructions: https://github.com/harman-f/MHI2_US_POG11_K5186_1-MU1476-AIO

No issues!

xirixiz, 2023-12-19:
@vinnyvinegar don't mention it 😄. 

Good to hear that you were able to solve it like this. Does it mean you're also running newer firmware/software now (K5186 instead of K3300)?

vinnyvinegar, 2023-12-20:
> @vinnyvinegar don't mention it 😄.
> 
> Good to hear that you were able to solve it like this. Does it mean you're also running newer firmware/software now (K5186 instead of K3300)?

@xirixiz Indeed! I have a 'volume level' now! It's nice to know how loud the radio is :D

There are probably other additions but that was the only feature I noticed almost immediately.

NittanySeaLion, 2024-01-01:
Found this in another forum:

"One quick question I did have, my 2021 came with MHI2_US_POG11_P5186 and after using the AIO with auto install enabled and all patches copied over, my head unit didn't flash to
MHI2_US_POG11_K5186. Is there something I missed?
The difference between P and K:

P = Production
K = Kunde (german for customer)

P FW are normally coming from stock/factory
K FW is installed for the customer (by dealer)

IIRC the script recognizes that the firmware levels are the same so the version does not change."