Mr-MIBonk/M.I.B._More-Incredible-Bash #372: MHI2_US_AU57x_K3341_MU1426_PATCH ifs issue

mkizito76, 2023-12-17:
**Describe the bug**
First of all, THANK YOU for the incredible work and assistance.
I was trying to an upgrade from MHI2_US_AUG11_S0533 as to MHI2_US_AU57x_K3341_MU1426 as described [here](https://www.audizine.com/forum/showthread.php/860600-C7-5-2016-A6-(and-others)-CarPlay-Android-Auto-Activation-DIY-Guide?p=14905268&viewfull=1#post14905268).
1. Here is the screen shot of my current FW.
[image]

2. Here is the latest map version I can get (has not changed)
[image]
3. I checked the REM and noticed the version had not changed so I tried running the update again (I assumed something went wrong) and this time I got the following error. Probably was a bad idea to do so.
[image]
It is the same error that was reported [here by Crazymax1990](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/discussions/82#discussioncomment-7871875)

4. I then cancelled the update and tried to apply the patch using the GEM menu and got an ifs error
[image]

5. I then checked the patches folder and noticed the .ifs file in the MHI2_US_AU57x_K3341_MU1426_PATCH patch is MU1426-ifs-root-part2-0x00b**a**0000-**2098EE00** but the system is asking for MU1439-ifs-root-part2-0x00b**e**0000-**0054e200**.
6. When I check the log file, here is what I find in it 
`Patch folder /net/mmx/fs/sda0/patches/MHI2_US_AU57x_K3341_MU1426_PATCH NOT found on SD card`
even though, the folder is there. Attached is a copy of the most recent log file.
[MU1426-JTB-07410.09.1511700F98.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/13697141/MU1426-JTB-07410.09.1511700F98.log)

Again, thank you for the assistance

R3DcooDe, 2023-12-17:
if you can find MHI2_US_AU57x_K3341 AIO use this one , long time ago i did this job from US AUG - I installed EUROPE AU57X AIO update and it worked great for me,  maybe it will works also for you

mkizito76, 2023-12-17:
> if you can find MHI2_US_AU57x_K3341 AIO use this one , long time ago i did this job from US AUG - I installed EUROPE AU57X AIO update and it worked great for me, maybe it will works also for you

In this case would I add this to the custom  metainfo2.txt file so the system sees it?

thanks for the quick reply

R3DcooDe, 2023-12-17:
nope dont change anything to metainfo-  just go to menu setup mmi and factory reset old system,  and after insert AIO update and install via red menu - standart ( no user define)  and i hope it will work great ..    if after final reboot last one unit goes down and take a lot of time to come back up -  just wait 2 minutese and press buton to open display up and it will be fine i hope so :>D   good luck

mkizito76, 2023-12-17:
> MHI2_US_AU57x_K3341
OK so ...
1. Reset system to original 
2. Insert AIO update on SD card (So all I have to do is fine this update). 
3. Use Red menu to update
Sorry for numerous questions.
So where does mib come in?

R3DcooDe, 2023-12-17:
no where you do not need mib for AIO updates - they have build in patch - please be careful with aio update for EUROPE it has for usa firmware i do not know if  AIO firmware exist

R3DcooDe, 2023-12-17:
one more question did you update from old firmware to latest one au57xx now you have problem with patch you cant install maps or activate audi smartphone interface or what you need

mkizito76, 2023-12-17:
> one more question did you update from old firmware to latest one au57xx now you have problem with patch you cant install maps or activate audi smartphone interface or what you need

I installed as described above. Copied mib to sim card etc... accessed GEM menu to do update but it would not.
I am trying to change FEC so I can have unlimited map updates.

R3DcooDe, 2023-12-17:
in your case i hope Iprot or Harman f are still alive and get back to you - as seem mib is not updated i long time now i think they abandoned this project :/

mkizito76, 2023-12-17:
Is there any issue doing a complete factory reset? I will try and find the version you suggested. 
thank you very much for helping

R3DcooDe, 2023-12-17:
for factory reset there is no problem but for update to aio i dont know if would any issue happens :/

mkizito76, 2023-12-17:
> for factory reset there is no problem but for update to aio i dont know if would any issue happens :/

I found this [video](https://www.youtube.com/watch?v=VS21Lyess7o). Will try it once I find the appropriate file

dekoderi, 2024-06-19:
> për rivendosjen e fabrikës nuk ka asnjë problem, por për përditësimin në aio nuk e di nëse do të ndodhë ndonjë problem :/

send me you contact number

Viibezz, 2024-09-04:
@mkizito76  Hello, were you able to get CarPlay with version MHI2_US_AUG11_S0533 at all?

Vadon1, 2025-07-02:
> > for factory reset there is no problem but for update to aio i dont know if would any issue happens :/
> 
> I found this [video](https://www.youtube.com/watch?v=VS21Lyess7o). Will try it once I find the appropriate file

hi. have you tried like on this video? it's working?