Mr-MIBonk/M.I.B._More-Incredible-Bash #374: Patch missing - MHI2_US_POG11_K3300

vinnyvinegar, 2023-12-18:
Hello, I am missing a patch folder corresponding to my MHI2 FW version, MHI2_US_POG11_K3300. I've uploaded the backup folder here: https://we.tl/t-wyQjtoqB8b

NittanySeaLion, 2024-01-01:
See issue #371 as well. I don't think there's anything to do. It looks like you ignore the "P" or the "K". From another forum,

"The difference between P and K:

P = Production
K = Kunde (german for customer)

P FW are normally coming from stock/factory
K FW is installed for the customer (by dealer)

IIRC the script recognizes that the firmware levels are the same so the version does not change."

pricootz, 2024-02-16:
Try installing the firmware **MHI2_US_POG11_P5186**

Fambroz93, 2024-02-25:
im having this same issue. Patch missing - MHI2_US_POG11_K3300. Any solutions to this?

Fambroz93, 2024-02-25:
Got it to work, all that needs to be done is rename MHI2_US_POG11_K3300 to MHI2_US_POG11_P3300

vinnyvinegar, 2024-02-25:
I installed the newer version here: https://github.com/harman-f/MHI2_US_POG11_K5186_1-MU1476-AIO

T3lion, 2024-03-11:
> Got it to work, all that needs to be done is rename MHI2_US_POG11_K3300 to MHI2_US_POG11_P3300

May I ask what you renamed, was it the folder or a file in the folder with that name?

Fambroz93, 2024-03-12:
It was just the folder that I had to rename, worked after that.

T3lion, 2024-03-12:
I will try again. Thank you for replying so quickly.

pavitardua, 2024-06-25:
> I installed the newer version here: https://github.com/harman-f/MHI2_US_POG11_K5186_1-MU1476-AIO

Hi I have the same firmware version as yours can you confirm the steps to successfully install Car Play with FW **MHI2_US_POG11_K3300 0809**

vinnyvinegar, 2024-06-25:
I just followed the directions here: https://github.com/harman-f/MHI2_US_POG11_K5186_1-MU1476-AIO?tab=readme-ov-file#how-to-install

It gave me some other new features too (like a volume meter)

pavitardua, 2024-06-26:
@vinnyvinegar  It did not work for me I got some errors and could not complete the process. The summary of devices a lot of them said NOK.

pavitardua, 2024-07-12:
@vinnyvinegar any suggestions, on how to proceed.

pavitardua, 2024-07-14:
@vinnyvinegar I was able to finally get through, thanks to the developers and you! 

It did not work the first time probably because the memory card I used was 8 years old, when I tried a brand new card it worked like a charm. The whole process took about 40 minutes but it was worth it!

JamesOakes01, 2026-10-24:
My solution was to rename the patch file MIH2_US_POG11_P3300_MU0809_PATCH to MIH2_US_POG11_K3300_MU0809_PATCH
I simply replaced the P to a K to reflect the firmware version in the car. I also reran the AIO thing and the carplay activation. The AIO showed a green screen that said "success" in big letters. After that and the carplay activation, I rebooted (the icon still hadn't appeared yet), I plugged in then it worked. I hope this helps a future person.