Mr-MIBonk/M.I.B._More-Incredible-Bash #379: Support for MHIS_US_POG11_P5251

mjgorman, 2024-01-09:
Hello,  I have a brand new 2024 Boxster with no CD drive. Seems they have put in a new software revision and when i tried to run the AIO package it failed with a message about metainfo2.txt signature is not correct. I'm assuming this is because this new software revision as the latest patch versions in any of the files seems to be 5186. 

[image]

Happy to help in any way I can.

pricootz, 2024-02-16:
Hey @mjgorman,

you have to install the firmware _**MHI2_US_POG11_P5186**_

mjgorman, 2024-02-16:
> Hey @mjgorman,
> 
> you have to install the firmware _**MHI2_US_POG11_P5186**_

I'm not aware of a way to downgrade. 

I went ahead and used the services of someone who manually logged in and enabled the FECs for everything I needed, but i'm sure lots of other people are still going to be in need of support for the newer 2024 firmware versions.

haimiko, 2024-12-29:
He @mjgorman.  I also have the same problem.    Did you connect a USB-->Ethernet adapter to the car then using a laptop login over network?   Can you please elaborate on what you did?

haimiko, 2024-12-29:
He @mjgorman.  I also have the same problem.    I'm able to login via telnet.  I tried enabling android auto via the MIB menu but it doesn't seem to do anything.