Mr-MIBonk/M.I.B._More-Incredible-Bash #382: Missing patch?

ashleyroberts3000, 2024-01-11:
Patch is in the patch folder but shows as missing?!? What am I doing wrong? Maybe change FW?

[image]

[image]

ashleyroberts3000, 2024-01-11:
[image]

Fambroz93, 2024-02-25:
Any solution to this? I am having the same issue.

ashleyroberts3000, 2024-02-25:
I had to use 7 zip to unzip the patch folder into the MIB patches folder. Then I had to rename the folder to change the name of my firmware patch folder from K3300 to P3300. Once I reran the AIO patch everything finally worked.

Fambroz93, 2024-02-25:
Hey Ahshley, im actually doing that right now and i think its working haha!!! Sent from my iPhone