Mr-MIBonk/M.I.B._More-Incredible-Bash #392: PCM 4.0 stuck at logo screen after MIB install and patch

VTEKKEN, 2024-01-20:
**Describe the bug**
HELP PLEASE!
I downloaded MIB and had it auto install (Swdlautorun.txt) into my MHI2, after that I was able to get into GEM and patched aa/cp. I then plugged my android device in and it was recognized as Carplay, however, no AA content appeared, but the AA logo was initially on. Then the AA logo also went away.  I then tried re patching AA/CP for POG24, however after doing that it rebooted and stayed in the Porsche logo screen, it is not taking any inputs at all.  I pulled the fuse, and the battery, but this still did not fix the issue. 

**Add M.I.B LOG file to your report**
[MU0984-JTB-07429.03.171192078A.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/13998819/MU0984-JTB-07429.03.171192078A.log)

**Screenshots**
[image]

**Additional context**
1) Unit is stock POG24, working well before, no know updates made to it prior to me owning it. 
2) Thing I did wrong that could have caused it after relooking at the instruction: SD card is a small one with adapter. 
3) I tried a full size SD with new image and was able to see that MIB did get reinstalled based on seeing the log and backup files filled. 
the Porsche logo stays on now even with the car off and locked, I'm keeping the fuse pulled to keep the battery from draining.

VTEKKEN, 2024-01-20:
I also used the backup function prior to patching AA/CP, so I do have the original backup files, but can not get to GEM where I can reload the stock images.

Cloudmania4, 2024-02-27:
Issue fixed?you have ax88772 based lan usb?

Cloudmania4, 2024-10-03:
Fixed?

cascoarc, 2024-10-07:
I haven't solved the problem...
I have the ax88772 board and Putty
What command should I use to solve the problem?

Thanks

Cloudmania4, 2024-10-07:
Can you enter the telegram group ?

cascoarc, 2024-10-07:
yes

cascoarc, 2024-10-07:
I'm already

cascoarc, 2024-10-07:
VAG-MIB2

cascoarc, 2024-10-07:
your nick on telegram?

cascoarc, 2024-10-07:
Yes

:

cascoarc, 2024-10-07:
Your nick on Telegram?

:

Cloudmania4, 2024-10-07:
Write in group hello

HansiNEA, 2024-10-07:
Could you add me too?
My nickname is HansiNEA
THANK YOU

cascoarc, 2025-08-22:
> have you fixed it?

Yes 👍

mv-go, 2026-03-05:
Hey! Same issue here. Freezes on the loading "Porsche" screen. It's a panamera. Tried accessing the file system via telnet and dlink dongle - no luck. Can't even ping the system. Tried rebooting and accessing via telnet immediately after the start - no luck. In adapter stats in windows it looks like there's "nothing" on the car end. The adapter and laptop are good - tried on Cayenne with mh2p - works fine. 5F looks operational via piwis, I can read the errors, do coding, etc. Any help?