Mr-MIBonk/M.I.B._More-Incredible-Bash #393: stuck in startup picture pls help 😭 image mod wrong name file..

6a4uHa, 2024-01-23:
the radio sound is still working, carplay music still works when phone connected,
it seems some error in display in unit screen and stunk in startup pic. even when I try to press MENU button to enter service menu.

Figured it out the path was from older versions

ERROR: (rs_file_open) Error opening "/net/mmx/fs/sda0/mod/images/VWG13_x452x/skin5/images.mcf.delta" for read: No such file or directory
'/net/mmx/mnt/app/eso/hmi/lsd/Resources/skin5/images.patch': No such file or directory
My files was **VWG13_P4521** from 3.6.0
But in 3.7.0 they called **VWG13_x452x**
[MU1367-HHA-07319.08.2070090113.log](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/14026846/MU1367-HHA-07319.08.2070090113.log)

what should I do to recover the unit , thanks!

6a4uHa, 2024-01-23:
> [https://drive.google.com/file/d/19Bi9Jw6qzP8MNQMmr_0VUX7FhWWEpnUr/view?usp=drivesdk](url)
Logs+backup
> [https://drive.google.com/file/d/19H2KAlKVx_4MEutE2fyaeafEtJksDsl5/view?usp=drivesdk](url)

`https://mibsolution.one/#/1/14/MU1367-MHI2_ER_VWG13_P4521_2-HHA-07319.08.2070090113`

R3DcooDe, 2024-01-23:
i hope someone will be able to help you but as far as i can see,  mib is abandoned - no one is updating mib anymore las update or when someone checked issue page was 7 month ago