Mr-MIBonk/M.I.B._More-Incredible-Bash #398: No patch for MHI2_JP_AU37x_P0902

pierceq, 2024-02-02:
Hi, there doesnt seem to be a patch for my audi S3 2018 with version number MHI2_JP_AU37x_P0902
Just looking to get RoW maps/nav.

Cheers.

R3DcooDe, 2024-02-02:
install mib make full backup and post to mibsolution and i hope someone will make a patch

pierceq, 2024-02-02:
Can do - I suppose its very hard to make my own patch from an existing one?

pierceq, 2024-02-05:
I saw that you can do a conversion to ER - noticed that there is an ER patch that upgrades from 
MHI2_ER_AU37x_P0902 (notice i just changed JP to ER as does the conversion)

so i thought aight ill do it myself and document what i did in case somebody else wants to do this:

This is my endeavour to use my japan import car's maps and nav in New Zealand 

so far, ive used 'europe region conversion' (the one thats its own menu thing) in MiB to ER 
then i apply MHI2_ER_AU37X_P5153 (MU: 1326) update in red menu
got an error version mismatch between ER and EU
to fix this i went into MiB GEM and switched region in settings to EU
now firmware is updating and worked.
Now I install mib gem 
Aio (All in one) patch au37x_p5089. but I changed the add fec txt file to have 023e00ee for ROW maps
After it reboots I go back to gem aio and add fec container
Also activate carplay - idk if I had to do these but did just in case

(Note at this point android auto works but I want the audi maps for my virtual dash :)) 

After that trying to install ROW maps gives me error that I am in Europe
I go into mib gem, change car variant region to ROW (not europe/row, this didn't work) and reboot
I then try install ROW maps but only get this far, with update grayed out
I assume this is because my fec didn't get added - (i dont know why it didnt add last time??), 
Then I add fec container
Now 023e00ee  code shows up on red menu.
And I can start map update!

closed 2024-02-05 by pierceq