Mr-MIBonk/M.I.B._More-Incredible-Bash #400: Problem with Patching MIHG with FEC

stn00, 2024-02-09:
I have a problem with installing patch to map. It seems that the FEC code is not installed properly.
My software version is MHIG_EU_AU_K1555. Installed soft with option SWDL, then installed M.I.B 3.7.1 and after patching there is no FEC code in the list and map is still blocked.

Could you give me a hint where to look for a solution?

[image]
[image]
[image]
[image]

matim23, 2024-03-14:
Have you found a solution? Im in the same situation. MHIG_EU_AU_K1555 
Trying to find 023000EE activation key so I can upload maps

aledepaoligampel, 2024-03-14:
I think that you need to use 021 instead 023. 021 is for MIB1 maps. 023 for MIB2 maps. In your case, you have a MHIG (MIB1High).

matim23, 2024-03-14:
Thank you so much. Since the website is down, is there any place I could possibly find them?
Or perhaps you could share one? :)

aledepaoligampel, 2024-03-14:
dont need a file. In the folder patch, you need to add or replace 023000EE to 021000EE code on the fec files (.txt)

matim23, 2024-03-14:
Good to know, I'll try then! 👍

MoNkToNk, 2024-03-18:
Did anyone solve this issue?

pemospec, 2024-04-16:
Same problem, anyone solved?