Mr-MIBonk/M.I.B._More-Incredible-Bash #403: Workaround for not listed patches, MHI2_ER_VWG11_P0255?

DiegelD, 2024-02-13:
Hello,

I want to flash this Repository SW to my MHI2, however its not working since I have an MHI2_ER_VWG11_P0255 and this patch in not in the patch compatibility table.
 
1. Are there any ways to work around this?
2. Something like a firmware update at the dealer? 
3. Or is this due to some Hardware related topic not possible?

[image]

Thank you for the support

pricootz, 2024-02-16:
Hey, @DiegelD. Look [here](https://mib-helper.com/index.php?train=MHI2_ER_VWG11_P0255#details)

You can upgrade your firmware first of all from **P0255** to **K3342**.

You can download the firmware from [here](https://mibsolution.one/) (ID and PASSWORD ar both **guest**)

And finally you can use M.I.B.

DiegelD, 2024-03-08:
Thank you,
look so far that it works. I can enter the green mode. Next week i will install the large display.

What i did:

1. downloaded the full patches. 
2. run the auto installer _Swdlautorun.txt to Swdlautorun.txt

closed 2024-03-08 by DiegelD