Mr-MIBonk/M.I.B._More-Incredible-Bash #410: Help with a patch please for MU0726-MHI2_US_POG11_P3276

Analytics-JDS, 2024-02-20:
### Discussed in https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/discussions/407

<div type='discussions-op-text'>

<sup>Originally posted by **Analytics-JDS** February 19, 2024</sup>
I have uploaded my backup folder and looking for help with a patch please.
MU0726-MHI2_US_POG11_P3276-JTB-07414.06.1611920F66</div>

xirixiz, 2024-02-21:
Didn't do this myself, but maybe better to upgrade to a newer firmware version:
https://mibsolution.one/#/1/9/MHI2%20-%20HARMAN/Firmware/Porsche

For Europe, I was already at the latest version (MHI2_ER_POG24_K5137_MU1417), so patching went fine.

Analytics-JDS, 2024-02-24:
Tried your suggestion, not other issues.

received the following error: Device reports error - RCC_Pre_Hashes.txt - ErrorCode119
multiple packages have [NOC]
No carplay enabled.

Firmware does seem to have been updated..

Analytics-JDS, 2024-02-26:
> Didn't do this myself, but maybe better to upgrade to a newer firmware version: https://mibsolution.one/#/1/9/MHI2%20-%20HARMAN/Firmware/Porsche
> 
> Patch is availble for the latest release for your car (MHI2_US_POG11_K5186_1_MU1476_PATCH)
> 
> For Europe, I was already at the latest version (MHI2_ER_POG24_K5137_MU1417), so patching went fine.
I went back and walked through this again. and this worked. Thank you.

closed 2024-02-26 by Analytics-JDS

xirixiz, 2024-02-26:
Late reply, but the weekend has been quiet busy. Good to read that it worked out!

sambam80, 2024-03-25:
I had the same FW version "MU0726-MHI2_US_POG11_P3276" but when I tried to update using "MHI2_US_POG11_K5186_1_MU1476_PATCH" I got the bleow during the update:

[image]
[image]

And now the PCM is restarting nonstop and can't seem to get past it

xirixiz, 2024-03-25:
Hi, but this looks old to be honest. Could you share some more details about your Porsche.
I own a Porsche Panamera G2, which has a different/newer interface then what is shown on the image you've shared.

Please share as much details as possible.

sambam80, 2024-03-25:
Mine is Porsche Macan 2017 Base model.