Mr-MIBonk/M.I.B._More-Incredible-Bash #413: I need help. Briked MMI. :'(

kcisoul, 2024-02-27:
I am encountering an issue with enabling CarPlay on my vehicle's MMI system, specifically the MHI2_KR_AU57x_S0035 firmware version. Despite successfully activating CarPlay and Android Auto, I'm unable to access these features due to the absence of the corresponding button in the UI, which lacks the updated Audi connect menu.

To address this, I attempted to switch the firmware to the European version (ER) by following a guide for converting to the ER menu. This conversion seemed successful, as the system's region changed to the European version. However, the smartphone audio interface option still did not appear in the menu.

Subsequently, I installed a patched firmware version from an external source, as recommended by a YouTube tutorial. Unfortunately, after this update, the MMI system now fails to shut down properly. Although I can occasionally access the main menu, (in case of battery  re-connection) which now includes the audio connect button, the system quickly goes black again (about 10 or 20 secs).

Considering these issues, is it possible to connect the MMI system with my MacBook via a D-Link connection to restore the original firmware or rectify this problem?

R3DcooDe, 2024-02-27:
well done mate, that's why you have to send to experts to do the job,   so where are you located to help to recover mmi

kcisoul, 2024-02-28:
I'm located in Seoul, Korea. :'(
I believe there are a few methods I can try to solve this issue:

Connect to the MMI via Telnet using the D-Link method and then try this command: "rm /mnt/efs-persist/FEC/mmx_fec_ids.bin"
Update the sound system parameter via VCDS.
I am currently waiting for a VCP cable and an AX88772 from AliExpress.
Could you offer any better suggestions?

Cloudmania4, 2024-03-19:
Why following youtube , what you did exactly , write what shit you installed

Cloudmania4, 2024-03-19:
Is stucking in bolo)

kcisoul, 2024-03-19:
I am reaching out for assistance regarding my Audi's MMI system, specifically version MHI2_KR_AU57x_S0035, which currently lacks the Audi Connect User interface. Despite enabling Apple CarPlay and Android Auto through the MIB shell, the menu remained inaccessible, prompting me to seek a solution.

To address this, I followed a recommendation to convert my MMI version to the European version and then proceeded with a firmware update. Initially, the system appeared to operate correctly. However, I encountered severe performance issues shortly after, leading me to believe there might be a problem. Regrettably, in an attempt to resolve the issue, I powered off the system, which resulted in it becoming non-functional (bricked).

Fortunately, I was able to replace the bricked unit with a used MMI under warranty service. Given my previous experience, I am apprehensive about attempting the process again without further guidance.

Could you please provide detailed instructions or recommend a course of action to safely update my MMI system and enable the desired features without risking another malfunction? Any assistance or insights you could offer would be greatly appreciated.

Thank you for your time and support.

Cloudmania4, 2024-03-19:
Telegram nickname please

kcisoul, 2024-03-19:
kcisoul. thank you~!