Mr-MIBonk/M.I.B._More-Incredible-Bash #424: Help, Audi A6C7 2015 Korea,   in stock : MHI2_KR_AU57X_S0035

vut73a, 2024-03-13:
Before installing the European firmware, I need to update the Korean MHI2_KR_AU57X_P0901 to the latest version, but there is a problem, I can’t find it anywhere, can someone help me, I will be very grateful? I can't install MIB, it won't install, best regards Vladimir

vut73a, 2024-03-13:
Where to download such firmware MHI2_KR_AU57X_P0901 ?

R3DcooDe, 2024-03-16:
hello if you didnt solve it yet contact me

vut73a, 2024-03-16:
We have not solved the problem .
I need to update to the latest Korean firmware
Do you have one?
Do you have watsapp?
My number is +380677045026 Vladimir

boxxeox1, 2024-03-20:
> 안녕하세요 아직 문제를 해결하지 못했다면 저에게 연락해주세요

> hello if you didnt solve it yet contact me

Hello can I get some help too?? , I'm looking for a firmware file (MHI2_KR_AU57X_P0901)

please Can I get it through my e-mail? rlarl15dnjs@naver.com

Thanks you :D

vut73a, 2024-03-21:
> > 안녕하세요 아직 문제를 해결하지 못했다면 저에게 연락해주세요
> 
> > здравствуйте, если вы еще не решили проблему, свяжитесь со мной
> 
> Здравствуйте, можно мне тоже помочь?? , ищу файл прошивки (MHI2_KR_AU57X_P0901)
> 
> пожалуйста, могу ли я получить его по электронной почте? [rlarl15dnjs@naver.com](mailto:rlarl15dnjs@naver.com)
> 
> Спасибо тебе :D

I don't have it, do I want to find it myself?

boxxeox1, 2024-03-21:
> I don't have it, do I want to find it myself?

> I don't have it, do I want to find it myself?

Thank you so much!!
 I couldn't find it even though I searched....

vut73a, 2024-03-21:
> > > > 안녕하세요 아직 문제를 해결하지 못했다면 저에게 연락해주세요
> > > 
> > > 
> > > > здравствуйте, если вы еще не решили проблему, свяжитесь со мной
> > > 
> > > 
> > > Здравствуйте, можно мне тоже помочь?? , ищу файл прошивки (MHI2_KR_AU57X_P0901), 
> > > пожалуйста, могу ли я получить его по электронной почте? [rlarl15dnjs@naver.com](mailto:rlarl15dnjs@naver.com) 
> > > Спасибо тебе :D
> > 
> > 
> > У меня его нет, хочу ли я найти его сам?
> 
> > > > 안녕하세요 아직 문제를 해결하지 못했다면 저에게 연락해주세요
> > > 
> > > 
> > > > здравствуйте, если вы еще не решили проблему, свяжитесь со мной
> > > 
> > > 
> > > Здравствуйте, можно мне тоже помочь?? , ищу файл прошивки (MHI2_KR_AU57X_P0901), 
> > > пожалуйста, могу ли я получить его по электронной почте? [rlarl15dnjs@naver.com](mailto:rlarl15dnjs@naver.com) 
> > > Спасибо тебе :D
> > 
> > 
> > У меня его нет, хочу ли я найти его сам?
> 
> Большое спасибо!! Я не нашел, хоть и искал....

When you find it, share it with me?

ziwon0, 2024-08-20:
Is the problem solved?
MHI2_KR_AU57X_P0901 File not found 
I'm in Korea and I'm trying to upgrade CarPlay. Can anyone please help me?

vut73a, 2024-08-20:
No, it has not.

lushnikaj, 2024-11-09:
Hello has anyone found where can we download the  MHI2_KR_AU57X_P0901 firmware ?

vut73a, 2024-11-10:
> Hello has anyone found where can we download the MHI2_KR_AU57X_P0901 firmware ?

No

vut73a, 2024-11-10:
> Здравствуйте, кто-нибудь нашел, где можно скачать прошивку MHI2_KR_AU57X_P0901?

No