Mr-MIBonk/M.I.B._More-Incredible-Bash #425: Audi A4 B9 MHI2Q US to EU with success...but problem with frequency FM, when car is start run

VAGserwis, 2024-03-13:
Hi all ;)
I do with success conversion MHI2Q from USA to EU, all working property , maps, EU language etc, all good...but problem with frequency FM....I try many times "save" radio program from 1 to 6, and add to my favourite...but, when I turn off car, wait around 10 min, frequency automatic come back to 87,5...and I need manualy select my favourite radio, MIB2Q dont remember the last radio station..maybe someone know where is mistake ? my traint now is mhi2q_er_aug22_p5092
Best Regards

closed 2024-03-15 by VAGserwis

Avantgardt, 2024-03-31:
Good afternoon, did you solve this problem?

traycho, 2024-03-31:
Same issue, when I click scan they appear again but only 87.5 on startup.

traycho, 2024-04-19:
@VAGserwis  I have same issue , is this resolved, it seems I am missing even the scan function, only seek available from the radio.

Gruby1983, 2024-09-30:
> Cześć wszystkim ;) Udało mi się przekonwertować MHI2Q z USA na UE, wszystko działa, mapy, język UE itd., wszystko ok... ale mam problem z częstotliwością FM... Wielokrotnie próbowałem „zapisać” programy radiowe od 1 do 6 i dodać do ulubionych... ale kiedy wyłączam samochód, czekam około 10 minut, częstotliwość automatycznie wraca do 87,5... i muszę ręcznie wybrać moje ulubione radio, MIB2Q nie pamięta ostatniej stacji radiowej... może ktoś wie gdzie jest błąd? Mój pociąg teraz nazywa się mhi2q_er_aug22_p5092 Pozdrawiam

Hi explain me step by step how to conwert from us to EU please 
[image]