Mr-MIBonk/M.I.B._More-Incredible-Bash #426: Problem with mibsolution site

matim23, 2024-03-14:
Hi there!
The website has a significant issue. Nothing works despite my repeated attempts using several web engines.
A file called 20230926_MIB_Patches.7z is actually mandatory for me. Is there someone who could share it?

I apologize for any trouble caused, I'm new to this forum.
Thank you in advance. 
Cheers!

vut73a, 2024-03-14:
email?

matim23, 2024-03-14:
> email?

kodowaniemateusz@gmail.com

vut73a, 2024-03-14:
ok

vut73a, 2024-03-14:
https://drive.google.com/file/d/1wTINSKGeTVVMU-a2ODsFqcpUVSB97LQh/view?usp=sharing

vut73a, 2024-03-14:
?

matim23, 2024-03-14:
I requested acces

vut73a, 2024-03-14:
yes, normal?

matim23, 2024-03-14:
Yes. thank you!

closed 2024-03-14 by matim23

vut73a, 2024-03-14:
oupen?

ludo5962, 2024-06-05:
bonjour le fichier est il le dernier en dates si non avais vous un fichier plus récent merci.

vut73a, 2024-06-05:
У меня его нет, сам ищу, может он у вас есть?

engineerhasanbor, 2025-05-04:
Hello guys, I've checked the website and It was not on right now. I need a patch for MHI2_ER_AUG22_P3346 can someone share it for me ?

vut73a, 2025-05-05:
> Привет, ребята, я проверил сайт, и сейчас его там нет. Мне нужен патч для MHI2_ER_AUG22_P3346, может кто-нибудь поделиться им для меня?

https://drive.google.com/file/d/1wTINSKGeTVVMU-a2ODsFqcpUVSB97LQh/view?usp=drivesdk

richiekaby, 2025-05-07:
Hello guys, I've checked and the website was not accessible. Looking for firmware update my vw golf mk7 either MHIG_JP_VW_K0415 or MHIG_JP_VW_K0336

domek2222, 2025-05-09:
Hello guys, I've checked and the website was not accessible.  Looking for firmware update.  I need a patch for mhi2q er aug22 p5092 mu1329.  I don't have the code 23000ee.

Lifeopti, 2025-05-10:
Hi guys

Does any have this file and want to share

MSTD_EU_SE_P5310

Rami916, 2025-05-28:
Hi
I have an ateca 2023 manual my sofware is on 0841 and i don't have a seat connect cause i leave in algeria
Please halp me

Rami916, 2025-11-04:
Does any one have these file for update
MOI3_EU_SE_R8700L