Mr-MIBonk/M.I.B._More-Incredible-Bash #431: Help with update version

matim23, 2024-03-17:
Hello, 
I would like to perform an EU conversion on my AUDI A4 B9 with MHI2_US_AUG22_P3252 MU software: 0692.
Would it be possible to install the MHI2_ER_AUG22_P3241 MU software: 0678 without bricking the system? I know it's a downgrade, but I can't seem to locate an equal or even newer version.
Thanks in advance

vut73a, 2024-03-17:
}-     MHI2_US_AUG22_K3347    }-      MHI2_ER_AUG22_P3346

matim23, 2024-03-17:
Спасибо! :)

matim23, 2024-03-17:
Спасибо! :)

closed 2024-03-17 by matim23

qf512415007, 2024-06-10:
谁知道MIB皮肤更换教程嘛？

Petrit95, 2024-06-18:
Hello i have this please can you halp me i wanna convert from USA-EUROPE MY SOFTWARE IS  (MHI2_US_AUG22_P3252)(MU0692)

[image]

vut73a, 2024-06-18:
> Hello i have this please can you halp me i wanna convert from USA-EUROPE MY SOFTWARE IS (MHI2_US_AUG22_P3252)(MU0692) [...]

MHI2_ER_AUG22_P3346 ваша последняя прошивка для Европы!
 Вы должны скачать её с mibsolution.one

vut73a, 2024-06-18:
> Hello i have this please can you halp me i wanna convert from USA-EUROPE MY SOFTWARE IS (MHI2_US_AUG22_P3252)(MU0692) [...]
[image]

Petrit95, 2024-06-19:
Also i have bose system do i need to update afer that 

Кроме того, у меня есть система bose, нужно ли мне обновить ее?

Petrit95, 2024-06-19:
[image]

Petrit95, 2024-06-19:
I donweload the same MHI2_ER_AUG22_P3346 
 But this is the typeof message

vut73a, 2024-06-19:
Bose обновлять было не нужно, после обновления прошивки с Bose  получается конфликт

Petrit95, 2024-06-19:
Возможно, другая прошивка в Европу для обновления прошивки

vut73a, 2024-06-20:
Последняя для Америки 3347, для Европы будет 3346 -100%
Вам нужно было обновить сначала до последней Америки 3347, а потом конвертировать в Европу 3346

vut73a, 2024-06-20:
> Возможно, другая прошивка в Европу для обновления прошивки

Последняя для Америки 3347, для Европы будет 3346 -100%
Вам нужно было обновить сначала до последней Америки 3347, а потом конвертировать в Европу 3346 .
Я надеюсь вы так делали?

Petrit95, 2024-06-20:
Я успешно обновил прошивку до Европы, но теперь у меня есть проблема с картами, не могли бы вы немного подтолить меня, я исправляю программное обеспечение и все, кроме сообщения о конфликте навигации

vut73a, 2024-06-20:
У вас есть Whatsapp?

vut73a, 2024-06-20:
> Я успешно обновил прошивку до Европы, но теперь у меня есть проблема с картами, не могли бы вы немного подтолить меня, я исправляю программное обеспечение и все, кроме сообщения о конфликте навигации

У вас есть Whatsapp или viber?

Petrit95, 2024-06-20:
Да, у меня есть whatsaap

0038971307060 напишите мне, пожалуйста

Petrit95, 2024-06-20:
привет по вашей ссылке на карту, которую вы мне прислали, я загружаю и помещаю ее на SD-карту (P350), после этого я ставлю и начинаю обновляться, но все равно (Error communicating with device Speech_Check), возможно, карта неправильно установлена ​​на SD

[image]

[image]

[image]

[image]

[image]

[image]