Mr-MIBonk/M.I.B._More-Incredible-Bash #436: Subwoofer Knob on MHI2Q_US_AUG22_P5087 MU 1316

CallMeWZRD, 2024-03-21:
Hey everyone, looking for a bit of support with enabling the Subwoofer Knob on my MIB2

I have recently retrofit a VC into my 2017 Audi A4 without B&O and had to also replace the MIB to support the maps feature on the VC. 

I have installed MIB2 with part number 8W5 035 880C and has firmware MHI2Q_US_AUG22_P5087, after succesfully patching and changing the dataset in 5f to fix sound issues I attempted to enable the Subwoofer Dial from GEM and it doesn't seem to work. 

The script runs and completes succesfully but it doesn't show the dial in the sound settings. 

Would anybody know why this is happening? Is it due to the firmware version?

Also, if i were to update the firmware on this MIB2 unit, will iIhave to upload the dataset once again to fix sound issues? I'm not able to find the datset file online and I'm very unfamiliar with making it myself. If anyone has it I would really appreciate it!

Thanks everyone!

CallMeWZRD, 2024-03-21:
Okay so I ended up finding a solution. 

It seems like the 3.7.1 M.I.B version was the issue. I used version 3.5.1 "Spring Edition", enabled the subwoofer knob and it stuck. 

However if anybody has the correct dataset for my vehicle and can provide it to me I'd really appreciate it!

Tomt97, 2024-09-06:
hi @CallMeWZRD how did you patch and change your dataset in 5f?  I have the same issues.   my old deck didn't work., retro fit a S5 w B&O into my Q5 without.   Sound is not like my old Q5.   Need to update my dataset from 20 to 30

I capture the old unit setting and new unit setting 

[image]
[image]