Mr-MIBonk/M.I.B._More-Incredible-Bash #450: Porsche US TO EU MAPS INSTALL PROBLEM

lanse45, 2024-04-12:
Hi!

My car is the 2017 Porsche Cayenne, American version. Software version was MHI2_US_POG11_P3300. I wanted to convert it to the European version. I installed MIB. I made train info to ER. So, software version is MHI2_ER_POG11_K5126.

I enable navigation to EU, format NavDB and patched unit.

I download "P350_N60S5MIBH3_EU.7z" file, format sd and transfer files to sd. But when I am trying install maps, at update screen saying "MIB2 navigation database, loading metadata". I am wating, but nothing happening. So I can't install EU maps. 

What I am making wrong, what I must do? I want install EU maps to my system. Thank you for answers.

[MU0809-LOG.txt](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/14960704/MU0809-LOG.txt)
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

adasxdm, 2024-04-30:
Hello, I have same problem… Did you manage to solve it? Thanks!

lanse45, 2024-07-01:
No, I couldn’t manage to solve the problem still.

adasxdm, 2024-07-01:
I already figured it out. Reinstall your firmware and be sure to select Speech and Language options. Maps update does not want proceed installing because of missing SpeechDB.

lanse45, 2024-07-01:
I tried update the firmware. I extracted the “MHI2_ER_POG11_K5126_1_AIO_MU1394_20230320.7z” file to the SD card.

When I inserted the SD card into the car and went to the Swdl menu, the update started automatically. I couldn't make the Bose system "N". There is no sound right now.

Can I bring back the sound and how? Thank you for your help!

edgarsvasilevskis63-alt, 2026-11-08:
I’m having the exact same issue. When I try to install the maps through the red menu, the update gets stuck on “Loading metadata.” I tried reinstalling the entire firmware using “User Defined” mode and selected every option except Bose. The update took over an hour. I also installed all the activation keys, but the map update still gets stuck on the same screen.

Does anyone know what else I could try?

adasxdm, 2026-11-08:
Did you load EU language dataset? ( M.I.B. -> Multimedia System -> Languages ). Load the European language dataset, then try to install maps once again. If it still fails to install, reinstall firmware once again. Maps update gets stuck on “Loading metadata” because off missing speech database and language data.  Also, if radio menu gets buggy (for example, you cannot manually set radio frequency), do a factory reset (SET - System settings - Factory settings or Vehicle handover) . Good luck.