Mr-MIBonk/M.I.B._More-Incredible-Bash #452: Fec MHIG K1555 problem

pemospec, 2024-04-16:
Hello,
i have problem with flashing fecs. After making aio flash i was thinking there will be fec for navigation but there is not. I tried also tu use function to flash from .txt but i get errors? And still no have fec to update navigation and even use it. I cant upload backup folder because it is to big. What i should do? 
[image]
[image]
[image]

Crashsm, 2024-05-09:
Hi, i see you have MHIG with K1555 (same as me) i assume it is in Audi A3 (by picture).
I successfully installed M.I.B. but i can not access GEM (green menu) any advice? 

Yes SD card is in slot 1, didn't even remove it from slot since installation.

It would be good to see if it works form me or i will get stuck at same problem. Maybe we can work out issue together.

Crashsm, 2024-05-12:
It WORKS! 
Steps i did:
1. update firmware to MHIG_EU_AU_K1555
2.  cleared error after firmware update with [I'm So XORy](http://mib-helper.com/im-so-xory/) - update confirmation
3. fleshed M.I.B  Pre - 3.7.1 (**pre-release version**).
4. Flashed patched image (didn't add any FEC containers on SD just flashed with 3.7.1. as downloaded)
5. rebooted unit and MAPS key was added - legal :)
6. downloaded 2024 EU maps - update of maps lasted ~~45 minutes 

All this was done on MHIG (MIB1) car is Audi A3 2015 (g-tron) but should be same for any Audi with MHIG (MIB1)

TIP i struggled to enter GEM (green menu) there is nowhere written key combination
for RED press back + upper left key
for GEM press back + lower left key
you must be on main menu screen when pressing combination and you must press same time both buttons it took me 5-6 attempts

JohnnyCSilva, 2024-05-24:
I have one question, I don't have the obdeleven, so am I able to see if there's any errors?
Where do I find the MIB 3.7.1? I can only find the 3.7.0

lprot, 2024-05-26:
> I have one question, I don't have the obdeleven, so am I able to see if there's any errors? Where do I find the MIB 3.7.1? I can only find the 3.7.0

Use this one as it has some MHIG bugs fixed https://github.com/lprot/M.I.B._More-Incredible-Bash/archive/refs/heads/main.zip

Crashsm, 2024-05-26:
i used pre-release version
here is link for all releases: https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/releases