Mr-MIBonk/M.I.B._More-Incredible-Bash #466: MHIG_JP_AU_S0271 - AIO Patching error

iFuckEggs, 2024-05-21:
I have a Japanese import A3 and I managed to get MIB installed with no issues, however when going to patch the unit to take FEC codes and EU maps I get greeted with this error.
[image]

I've attached the log file. 
[LOGS.txt](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/15393818/LOGS.txt)

I can get the FEC code to accept and show as legal, however as soon as the unit is powered off and back on it then shows as illegal as it's not patched. 

Is this due to the Japanese firmware? I've noticed it's a little outdated as the latest is SO274. Would updating the firmware help or would I need to completely change to EU firmware? I'd like to avoid the latter if possible as I couldn't find much info on it for the older MHIG units.

lprot, 2024-05-26:
Did you try to update to MHIG_JP_AU_K0277?

iFuckEggs, 2024-05-26:
> Did you try to update to MHIG_JP_AU_K0277?

Yep, updated to K0277 and get the same error but with a bit more detail. 

[image]