Mr-MIBonk/M.I.B._More-Incredible-Bash #471: Header seize on unit does not match patch files size

guizmo84170, 2024-05-26:
Hi, on my 2014 Audi TT MK3 I have an error when I try to patch in 5098: Header size on unit does not match patch files size.
So it's impossible to activate carplay.

I had several errors when updating the firmware to 5098, I tested several files, several SDs without success. if anyone can help me...
[image]
[MU1339-LOG.txt](https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/files/15448094/MU1339-LOG.txt)

artec12, 2024-12-05:
Hello. I have the same problem. Did you find any solutions?

guizmo84170, 2024-12-05:
yes it's your firmware which is not properly installed in 5098, you must
have NOK, try again to install the firmware, the MQB and LTE were missed
for me.

artec12, 2024-12-05:
Thanks a LOT for your feedback! So I just have to install 5098 again and then install MIB again and run the patch and it should work.
What is NOK? Is there a way to tell if everything is installed correctly this time?

guizmo84170, 2024-12-05:
yes that's it. When you update your MMI you should find yourself in the
device summary with the list of all the car's devices. After the update you
have 3 counters: OK/NA/NOK

you must not have a NOK at the end. When your MMI restarts at the end of
the update you have the summary of the devices with OK-Y-S etc. opposite
and you must not have a NOK, otherwise that means that the update is going
badly. The error you get when you run the script means that the size of
your firmware does not correspond to the size of the patched firmware and
that there was therefore a problem during the MMI update.

What basic version of MMI did you start with? 0320? because it is often a
source of problems

ruts-code, 2024-12-05:
Hi,

I faced today with the same error when was trying to apply AIO patch. Just to confirm if I also should follow SW reinstall (current version was installed in AUDI center, but I assume they did some path as well because adapt my car from US to EU)

Another question I have if 023000 FEC key will be reinstalled together with new SW, because currently my FEC key gone(

[MU1329-YJL-26005.04.1790390159.log](https://github.com/user-attachments/files/18030810/MU1329-YJL-26005.04.1790390159.log)

guizmo84170, 2024-12-06:
Hello, same problem, files size doesn't match, so for me your upgrade P5092
is not perfectly installed.
FEC 23000ee (lifetiem GPS) will be install with MIB script

ruts-code, 2024-12-06:
Thank you for reply. That's what I plan to do.
The only question I have if is any specific steps for overwriting SW the same version? Because I will install  P5092 version over the latest available P5092 on Mibsolution.

guizmo84170, 2024-12-06:
You must reflash with same firmware or with an update. You can't downgrade
😢

artec12, 2024-12-09:
I tried to reflash the same version as advised and I got the following message "downgrades not selected for download found". Any idea what that means?

ruts-code, 2024-12-14:
I tried to update the same version, but system did not allow even starting update. It saw all modules as those which are not required update.

[image]
[image]
[image]

guizmo84170, 2024-12-14:
What is your car ?

ruts-code, 2024-12-14:
Audi Q5, 2018 (US version).
It was adapted to EU in Audi service station about 3 years ago.

guizmo84170, 2024-12-14:
Ok, i will check this evening, i have an idea of your pb (sorry for my nad
english, i'm french)

artec12, 2024-12-14:
> […](#)

I got this message while trying to reinstall the update. Dis you also get it? "downgrades not selected for download found".

ruts-code, 2024-12-15:
I found a solution for installing the same SW version. You need to allow custom SW update in M.I.B. After this, you will have an option to choose custom update in red menu and select all components to override them.

After SW overwritten AIO patch applied.

guizmo84170, 2024-12-16:
Great news ! Can you send me your firmware please or make a link . I want
to but a Q5 in few weeks

standrun, 2025-01-12:
anyone solve the problem?
I have reinstalled the same SW (everything is ok).
 I reinstall M.I.B but the problem is still here

guizmo84170, 2025-01-13:
What is your Fw version?

standrun, 2025-01-13:
> […](#)

MHI2 CN AUG22_K0730_MU0729.
I update FW from MU0728

guizmo84170, 2025-01-13:
Do you have this firmware in the patch folder on your MIB SD card ?

standrun, 2025-01-13:
> […](#)

yes. I edit the path name as requesrtd.
I tried other features and it worked