Mr-MIBonk/M.I.B._More-Incredible-Bash #478: 1033 Argo Tunnel error when accessing website

orcoaster, 2024-06-03:
Patched and updated my B9 S5 before the weekend, and thank you for all the work, it's fantastic and I am now on 2024/2025 maps.
I realized that I hadn't downloaded the current MMI software update or the VC update, and wanted to look for updates for the B8.5 S4 as well, but cannot access the website.  When I try the URL it returns a 1033 Argo Tunnel Cloudflare error.
I don't have experience with Cloudflare, so I don't know if that's just a safety measure if I had the site open in windows in both Firefox and Chrome, or just something it does time to time.  

Again, thanks for the hard work, the toolbox is fantastic.

ludo5962, 2024-06-03:
bonjour a tous moi pareil depuis hier impossible d'accéder a mibsolution même erreur 1033 Argo Tunnel Cloudflare je pense que sa vient du site

ludo5962, 2024-06-03:
sinon quelqu'un aurais le fichier zip de MIB_Patches

eskimanasse, 2024-06-04:
I do have the same problem.
Is that mibsolution.one still a valid page?

orcoaster, 2024-06-04:
It was as of last Friday when I downloaded a file.

mrmtb, 2024-06-04:
When I checked the site on Saturday, I was also receiving the argo tunnel error.

mcowell, 2024-06-05:
Error still exists as of today.

ludo5962, 2024-06-05:
Bonjour a l'heure actuelle toujours pas d'accès.besoins du fichier M.I.B_Patches .7z si quelqu'un aurait un lien pour cela merci .

ludo5962, 2024-06-06:
bonjour alors je suis arrivé a me connecter a MIBsolution par contre pour les fichier mibsoluion/mih2 il est vide

eskimanasse, 2024-06-07:
Yes, now it seem to work again!
Oui, ca marche bien!

Kiwy-max, 2024-06-08:
Hi,

mibsolution.one does not seem to work for me. Anyone else has the same issue.

The error is:

Database error: SQLSTATE[08004] [1040] Too many connections

Thanks :)

ludo5962, 2024-06-08:
Oui effectivement trop de connexion

SergDi, 2024-06-10:
the same 
Database error: SQLSTATE[08004] [1040] Too many connections