Mr-MIBonk/M.I.B._More-Incredible-Bash #485: Problem with mibsolution site - Latest official firmware update

Shoier9, 2024-06-11:
Hi there!
Can anyone send me the latest official firmware update (MHI2_ER_VWG13_K4525) of mib 2.5. Now I have MHI2_ER_VWG13_P4512_1

I apologize for any trouble caused, I'm new to this forum.
Thank you in advance.
Cheers!

Bince92, 2024-06-12:
Hello as someone the firmware MST2_EU_SE_ZR_P0516T? or as a clue where to find a proper download site?

razvy80ro, 2024-09-02:
Hi there!
Can anyone send me the latest official firmware update (MHI2_ER_VWG13_K4525) of mib 2.5. Now I have MHI2_ER_VWG13_P4512_1

Thank you in advance.
Best Regards!

iancoop, 2025-05-17:
Hello @Shoier9, @razvy80ro, did either of you ever get hold of the MHI2_ER_VWG13_K4525 firmware ? I am looking for it also, would be very appreciative if someone can share it with me.

Thanks !

asamofal, 2025-06-07:
`MHI2_ER_VWG13_K4525` (and many other versions) can be downloaded from there:

[https://mibsolution.one/#/1/9/MHI2%20-%20HARMAN/Firmware/VW](https://mibsolution.one/#/1/9/MHI2%20-%20HARMAN/Firmware/VW)

login/password: guest/guest

kbundg, 2025-11-16:
> i have a MHI2_ER_VWG13_K4525_1_AIO_MU1367_20230318 I can make torrent file

This would be hugely appreciated! ATM I can't seem to get permissions to download from mibsolution.one