Mr-MIBonk/M.I.B._More-Incredible-Bash #486: mibsolution.one: Database error: SQLSTATE[08004] [1040] Too many connections

eskimanasse, 2024-06-11:
Hi,

can anyone correct the error in https://mibsolution.one/ ?

There is a connection error, which gives an error message:
Database error: SQLSTATE[08004] [1040] Too many connections

Cannot log in.

Maybe reset is needed?

Rasho13, 2024-06-12:
agreed, still not working as of yesterday.

CuccioloDiInformatico, 2024-06-12:
Same problem today

Rasho13, 2024-06-12:
now it states "file not found."

CuccioloDiInformatico, 2024-06-12:
> now it states "file not found."

For me always the database error

arjayfon, 2024-06-13:
I have tried a number of different mibsolution.one urls and yesterday got the too many connections error.  Today I get the file not found.  I presume this is still a valid solutions program.

Rasho13, 2024-06-13:
seems to load today , managed to login but no files.

ludo5962, 2024-06-13:
Apparemment les codes de connexion ont changé

Rasho13, 2024-06-13:
Got in touch with them, site is undergoing maintanance due to switching accounts because the memory is full or something. Process takes time but it ahould be fully operatational soon.

CuccioloDiInformatico, 2024-06-13:
For me it says invalid username but I write guest correctly

eskimanasse, 2024-06-13:
Yes,
Now the login is working, but guest user cannot log in...

Rasho13, 2024-06-13:
BE PATIENT.

Wittereus, 2024-06-17:
Any update available?

arjayfon, 2024-06-17:
I was able to access and download yesterday.   There is apparently a lot of traffic on the server and it took multiple efforts before the login was successful.  Don’t try just once, try repeatedly until the door opens.  Might take a while.