Mr-MIBonk/M.I.B._More-Incredible-Bash #498: MHI2_US_POG11_K3300 0809

pavitardua, 2024-06-25:
When I tried following the steps I see the following screens in the order I am pasting them. I had to click on Cancel backup documentation and come out of the setup as it did not go through. 

**Screenshots**
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

P.S I saw much later that my firmware version **MHI2_US_POG11_K3300 0809** was not listed in the Patch Compatibility Table. Does this mean it won't work for this version ever? Can I expect an update that supports this particular version?

**Add M.I.B LOG file to your report**
I don't see any logs in /backup/logs folder.

**Additional context**
Add any other context about the problem here. - Luckily I am able to still use the stock PCM of my Porsche and nothing has broken. Tested connecting an iphone after this, it says Ipod Connected so it confirms Car Play was not installed.
Was the unit worked on before? - Nope
Is the unit stock in car or a retrofit? - It's a stock unit

Apps215, 2024-07-15:
pavitardua did you ever get any resolution to this?

pavitardua, 2024-07-16:
Yes, I have used this repository and it's instructions https://github.com/harman-f/MHI2_US_POG11_K5186_1-MU1476-AIO which is compatibe with my firmware.

Also this video is helpful: https://youtu.be/QPRqR47_9qo?si=7SeiboEDCn55KaTA

Please follow this thread: https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/issues/374#issuecomment-2227302515

Note: Please ensure you are using a good SD card, I haven't use external power but have kept the car engine turned on for the entire length of the updates which was about 40 minutes. Let me know if you have more questions as I have recently done this.

closed 2024-07-16 by pavitardua

rd007000, 2024-09-19:
Hello pavitardua

The pictures above show you have a Macan like I do. The video you posted is for a Boxster.  Did you follow the same procedure for your Macan as in the video?

Thanks