Mr-MIBonk/M.I.B._More-Incredible-Bash #530: Audi A3 etron (2020) - Initalising Navigation System stuck at 11%

n1ckjansens, 2024-08-14:
### Discussed in https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/discussions/525

<div type='discussions-op-text'>

<sup>Originally posted by **williamsrh** August  6, 2024</sup>
Hello all,
I bought the car just over a month ago and since then I've had an issue with the Navigation System not initialising, 
[image]
I obtained a copy of the M.I.B software and the newest 2024 map version in the hope that this would solve the issue, 
I went through the installation instructions for installing the M.I.B software and applying the activation codes all of which seem to have been done successfully, 
[image]
[image]
However, I can't update the maps either through the online update in the car or using an SD card, I keep getting the error "No valid update data found on this device". 
[image]
For some reason in the version info panel, I don't have the navigation database option, 
[image]
After the last couple of weeks of watching YouTube videos and reading numerous threads. I thought I would reach out and see if anyone in the community could help me,
I'm happy to obtain any additional images or resources if that would help identify the issue, 
Thank you in advance, 
Rhys</div>

bwilliams75, 2024-08-25:
How were you able to get into the GEM? I am having the issue and can’t do anything.

n1ckjansens, 2024-08-27:
> How were you able to get into the GEM? I am having the issue and can’t do anything.

Actually i have full working MMI except Navigation :(

Jiherded, 2024-09-25:
I have the same problem, 11%). And  M.I.B. interface has become smaller.  The activation codes were not installed. But MMI is  full working MMI except Navigation. My firmware is MHI2_ER_AU37x_P5089.

chefranov, 2024-10-15:
Clear NavDB and re-install maps

bvass1, 2025-04-13:
Hi, I also have the issue with navigation loading to 11% but my version is MHI2_ER_AU37x_P5153
I also have the B&O sound, if I am updating how should I remove sound from the update?
Thanks in advance

RawBee4572, 2025-05-31:
@chefranov 

> Clear NavDB and re-install maps

I have a similar issue and clearing NavDB didn't help.

I've tried to reinstall all maps, I've tried to reinstall just some maps (by enabling user-defined installs + installing just some maps through REM).

It's really a very interesting issue as it evolves over time in a really weird way.
For me it started out of the blue. I've driven for some time with enabled maps, then suddenly this issue appeared. 

Accidentally I've noticed that if I turn off usb <-> wireless CarPlay then after 3 minutes of being stuck on 11% it will finally load maps. If CarPlay device is plugged in it never finishes and stays on 11% forever.

Another interesting thing I've noticed is that if early in the boot process I initiate REM then it opens REM but it is unresponsive for some time. And then REM becomes responsive AND maps suddenly appear. 

It's like some deadlock happens but triggering REM early forces it quit it.

Now... Based on hypothesis that it is an internal deadlock, I've made an assumption that maybe if I have only minimum maps installed (I had all of them) it'll work out. So I've did clear NavDB (unfortunately without any backup of NavDB prior).

It didn't help out. Right after that reset it shows "Navigation data can not be read" (expected), but then if I install all maps as before (installing just some of them unfortunately become extremely messy) I still get stuck on 11% and now it is way harder to get through even through REM (it works irregularly) and it isn't working at all even if CarPlay device is plugged out.

If anybody has any ideas what else can be done, it'll be greatly appreciated.

RawBee4572, 2025-06-01:
@n1ckjansens solution proposed in https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/discussions/448#discussioncomment-10648661 helped me, more specifically

1. Clear NavDB

2.  
```shell
rm -Rf /net/mmx/mnt/gecache/*
# This command recursively deletes everything inside /net/mmx/mnt/gecache/ without confirmation.

on -f mmx sync
# and then reboot
```

3. install maps again

Truffles directory was empty so removing it was not affecting anything
 
@chefranov @harman-f @Mr-MIBonk @lprot 
Maybe those steps can be added to `delnavdb` script, clearing `gecache` seems to be an appropriate thing to do after reformatting the whole nav partition.

giohouse, 2025-06-17:
How can you clear NavDB?

Sid8899, 2026-09-27:
Hi @n1ckjansens do you still have MHI2_ER_AU37X_P5153 file ?