Mr-MIBonk/M.I.B._More-Incredible-Bash #541: Several files are corrupt on mibsolution.one MOI3 section

wielspin, 2024-09-01:
Several files in the MOI3 section (MQB_Solution/MOI3/Preh) on MIBSolution are corrupt. They can't be uncompressed. Tried with Winrar and 7zip software.

MOI3_EU_VWMQB_P30600P_3G0919360QC.7z
MOI3_SEQREL_EU_VWMQB_P33002P_ 3G8919360S.zip
MOI3_SEQREL_EU_VWMQB_P33202P.7z

SoulRaven, 2024-11-23:
i confirm that all 3 files are corrupted

jogoossens, 2024-11-27:
Indeed, same thing here!

mastum, 2025-01-16:
Has anyone found another source for the download?
Thanks

SoulRaven, 2025-01-16:
i have upload the files back to MIBSolution each file contains a md5 hash.
for what i know the files are OK

mastum, 2025-01-16:
Thanks for your reply, I redownloaded the .rar archive but it is still corrupted and also the md5 does not match.

[image]

SoulRaven, 2025-01-16:
check from this, i my own space: https://mega.nz/fm/m3plzAJC

SoulRaven, 2025-01-16:
update the link

mastum, 2025-01-16:
Many thanks!

navidoctor, 2025-02-03:
If you send the partnumber is will upload it

regards

mehmetcomez, 2026-11-14:
@navidoctor — following up on your offer to upload firmware when given a part number.

Looking for the exact file:
**MOI3_RW_VWMQB_P30600P** (RW / Rest of World, Preh, MIB3 Option)

Unit details:
- Part number: **3G5035832A**
- Current train: MOI3_RW_VWMQB_P27000P / SW 0270
- Hardware: H22 Preh

Public mirrors (MIB Vault, mibsolution.one, Mega dumps) only have RW P27800P / P33000P / SEQREL — not this exact RW P30600P. Would you be able to upload it?

Thanks.