Mr-MIBonk/M.I.B._More-Incredible-Bash #548: Audi smartphone interface - no response when connect USB (Audi Q7 4M)

Umid9405, 2024-09-17:
Hi, I have Audi Q7 4M, I did everything as per your instruction, Audi smartphone interface activated. But when im trying to connect USB (I have 2 ports and 1 Aux, 1 port for iPod) no responses, i tried both AA and Carplay not working only message - no devise is connected which support the Audi smartphone interface etc. USB originals for Android and for Iphone working in other cars. 
FW: MHI2_ER_AUG22_K2161_MU1421

sardella-dav, 2024-10-09:
Hi, I'm experiencing the same exact issue on my Audi TT MK3, did you find a way to solve it?

me61ic, 2024-10-13:
And also another one experiencing the same. This time in a 2017 Audi A3 8V. 
MHI2_ER_AU37x_P5089 with software 1326 here.
Everything works but when I try to connect through the cable in either USB sockets, I can see my iphone (Carplay) under Audi smartphone in Connection Manager, but when I select it nothing happens in the phone, apart from the charging noise, no Carplay dialog or anything, the connection is trying to be stablished for a few seconds and the refused somehow, and deselects from the Connection Manager.
Video showing it:
https://youtu.be/KxkUxjJUhgQ?si=8BFjKsVVPfpOYE4y

Have any of you found a way to solve it?
Thanks in advance

Meyertrix, 2024-10-24:
Did you update the firmware on the Red Engineering menu? FW can be downloaded from https://mibsolution.one/#/1/9/MHI2%20-%20HARMAN/Firmware/Audi
It is a seperate memory card for FW and another memory card for MIB.
After FW upgrade install MIB again and from Green menu run the AIO patch.

eriloz83, 2024-10-30:
I have the same problem with my VW. After pluging in into the usb nothing happens. It does not recognize the phone. 
FW version: MHI2_ER_VWG11_K3342

hcg890, 2024-11-16:
What is the process for updating the firmware? Is it build in to the MiB main download? The 2017 Porsche Macan I tried this on did an update when I first put the SD card in, then I used the separate GEM patches update after. 

The car shows that CarPlay is installed but I received the message "No device connected - Connect your phone via USB to use Apple CarPlay". The usb port is still able to provide power.

Do you have any advice?

Meyertrix, 2024-11-19:
> What is the process for updating the firmware? Is it build in to the MiB main download? The 2017 Porsche Macan I tried this on did an update when I first put the SD card in, then I used the separate [...]

Yes I do...
You are one step ahead. Take another SD Card and goto https://mibsolution.one/#/1/9/MHI2%20-%20HARMAN/Firmware/Porsche 
Look for the right firmware for the 2017 Porsche Macan and download that onto the new SD Card. (you might need to do some research to confirm which patch to use.) 
Copy the patch files onto the new SD Card
Insert new SD Card into your car and from Red Engineering menu update the firmware following the steps provided here - https://youtu.be/KxkUxjJUhgQ?si=8BFjKsVVPfpOYE4y
Now that you have firmware which can take the patch updated on the car can you continue with the original SD Card and because you overwrote the firmware MIB has to be installed again.
Make sure the firmware you updated to has the correct (same) patch in the patches folder on the MIB SD Card.
Once installed you can now patch the system and android auto will work.

So in simple terms MIB can not patch your factory firmware. You need one of the modified firmwares from the Mibsolutions.one site and only once a compatible version is installed can MIB patch it to make Carplay work.

B3none, 2024-11-19:
@Meyertrix do you have a list of updated patches we can use? The latest release does not include the AUG24 cars as mentioned in #596

Meyertrix, 2024-11-19:
@B3none I dont have any firmware and can only direct you to the site to use what is available. You’ll need to contact the developer and upload a backup of your firmware for modification and patching.

Meyertrix, 2024-11-19:
> I have the same problem with my VW. After pluging into the usb nothing happens. It does not recognize the phone. Please help what to do. FW version: MHI2_ER_VWG11_K3342_MU1427

Do you have the right FEC codes in the Red Engineering menu
You need - 00060800 Google Android Auto and 00060800 Apple Carplay

eriloz83, 2024-11-19:
> > I have the same problem with my VW. After pluging into the usb nothing happens. It does not recognize the phone. Please help what to do. FW version: MHI2_ER_VWG11_K3342_MU1427
> 
> Do you have the right FEC codes in the Red Engineering menu You need - 00060800 Google Android Auto and 00060800 Apple Carplay
> 
> Edit: Is this your original firmware or did you update it from the site? See my previous comment maybe it will help you. I see the MHI2_ER_VWG11_K3342_MU1427 is in the list but not sure if you upgraded to it or if it is factory. If factory download the patched version from the site and update it on the car.

No I don't have that FEC. How can I install it?

eriloz83, 2024-11-19:
> Do you have the right FEC codes in the Red Engineering menu You need - 00060800 Google Android Auto and 00060800 Apple Carplay
>
> Edit: Is this your original firmware or did you update it from the [...]

It worked. I did everything again from the first step and now it is working. Previously I did it in wrong order... Big thank you!