Mr-MIBonk/M.I.B._More-Incredible-Bash #553: Telegram URL is Expired

speedythesnail, 2024-09-22:
Hello all,

I am unable to join the telegram group, the link is expired.

nirvananils, 2024-09-24:
Then you got blocked in group

speedythesnail, 2024-09-24:
> Then you got blocked in group

Would you check please? I have the same userid there as in here.

I don't know why I'd be blocked.

nirvananils, 2024-09-25:
No as iam not admin in there

speedythesnail, 2024-09-26:
All right, well I opened this issue because the link in the repo does not work [for me].

Whether I am blocked for some reason or not is up for debate, while I know you have no obligation to help me, if somebody is in this repo (whether a contributor or browser) would be able to reach out to the group to see if the link is valid, that would be wonderful.

This is a great project and I am thankful it exists and would love to be able to participate in the Telegram chat.

Thanks all.

krcenov, 2024-11-15:
Link is https://t.me/joinchat/EHt4RRksHcMQk6Xi6tFaBw