Mr-MIBonk/M.I.B._More-Incredible-Bash #564: unable to download content from mibsolution.one

ckirim1, 2024-10-06:
**Describe the bug**
A clear and concise description of what the bug is.

**Check known issues before reporting**
Make sure you have checked here:
https://github.com/Mr-MIBonk/M.I.B._More-Incredible-Bash/wiki/Prepare-M.I.B-SD-card-&-install-M.I.B#%EF%B8%8F-trouble-shooting-%EF%B8%8F

**Add M.I.B LOG file to your report**
Always attach your M.I.B LOG file!
Check M.I.B SD card in folder `/backup/` and upload that file together with your report.

**Screenshots**
If applicable, add screenshots to help explain your problem.
E. g. of the error message on screen.

**Additional context**
Add any other context about the problem here.
Was the unit worked on before?
Is the unit stock in car or a retrofit?

twitnic, 2024-10-11:
This - here - is GitHub. No mibsolution Issue Tracker

x10guy, 2024-11-13:
@Mr-MIBonk not to grab your attention, but I am unsure who owns the mibsolution site... but if they could be flagged that it is currently down (as well as the wiki) that would be awesome!