Mr-MIBonk/M.I.B._More-Incredible-Bash #586: Mibsolution Argo Tunnel Error

AzzaQ7, 2024-11-11:
Hi, I'm trying to get the latest update file for my Audi Q7 from mibsolution, Mib helper points me to it but I only get an error page with Argo Tunnel Error. My currant version is MHI2_ER_AUG22_S2140 and I am looking for MHI2_ER_AUG22_P3346. Any help would be appreciated.

Thanks.

mpurch, 2024-11-11:
https://mega.nz/file/cEliiSTa#8cYt17NbEbMLV-Kg47NvKbGeiF3h5HPIba9uxXlWKAA

May be the firmware you're looking for, its MHI2_ER_AUG22_K3346, Stopped at MHI2_ER_AUG22_K3344 as there was no difference on my Q7.

However I am unable to get the MHI2_ER_AUG22_K3344 patch files since mibsolution is also down so can't activate anything even though I have M.I.B now flashed correctly. So you'll have larger problems if the site doesn't come back online.

AzzaQ7, 2024-11-12:
Thanks for that, hopefully it comes back online.

snatimicale, 2025-04-02:
after this updateMHI2_ER_AUG22_K3346 the audio no longer works, can you help me?