Mr-MIBonk/M.I.B._More-Incredible-Bash #594: USB connection not found after updating Porsche Macan 2017 with PCM 4 to Car Play

hcg890, 2024-11-16:
I tried to install the software, but had some issues. 

I managed to install MIB and went through the green menu twice to install the patches. Looking at the logs in the backup folder, it seems to have installed CarPlay. 

There is an option for Smartphone link now, but I get the message "No device connected - Connect your phone via USB to use Apple CarPlay". 

I used two different cables and two phones iOS 18.1 to try the connection, but it still didn't work. The CarPlay option doesn't show up on the phone either. The USB cable is able to provide power to the phones via the port in the center console that is also for CarPlay.

Do you have any advice for this?

__________
Logs:
 __  __     ___           ___  
|  \/  |   |_ _|         | _ ) 
| |\/| |    | |          | _ \ 
|_|  |_|ore|___|ncredible|___/ash - V3.6.0 "QR Edition"
______________________________________________________________________
______________________________________________________________________
 -> done :-)
FEC folders...... -> done :-)
Shadow files..... -> done :-)
Dataset's........ -> done :-)
EEPROM...........
-----------
timestamp: 1970_01_01_00_02_10
 -> done :-)
storage1......... -> done :-)
storage2......... -> done :-)
backing up backup -> part 1 done
ls -als /net/mmx/mnt/boardbook --> total 192
      64 drwxrwxrwx  3 root      root          32768 Jan 01  1970 .
      64 drwxrwxr-x  5 root      root          32768 Jan 01  1970 ..
      64 drwxrwxrwx  3 root      root          32768 Jan 01  1970 MU0809
Free space on /net/mmx/mnt/boardbook 1012M

-----------
timestamp: 2024_11_15_16_58_46

RCC-launcher---->
-- Swdlautorun.txt disabled
-- Skip MOST enabled
-- Illegal and withdrawn FECs removed
-- Cleanup done!
AppDevelopment.jar found
-- GEM version OK - no update needed
-- M.I.B GEM link installed
-- Development Mode - GEM - activated

-----------
timestamp: 2024_11_15_17_08_59

RCC-carp---->
CarPlay and Android Auto activated now

-----------
timestamp: 2024_11_15_17_09_05

RCC-addimage---->
Installation finished :)

-----------
timestamp: 1970_01_01_00_09_15

MMX-flash---->

MMX-flash---->

MMX-flash---->
ifs-root-part2 (ASCII): 00f0
3 image type=2(IFS     )  ba0000...1a8a1ff (15639040)|93...213 (120):LOCKED
0x00ba0000 - 00A2EE00

Found JTB-07408.04.16119301DF Backup, will check if all complete...
Standard-Backup will require ~5 minutes to complete!
Partition info... -> done :-)
SWDL............. -> done :-)
Version info..... -> Backup is already there!
GEM info........ -> done :-)
Folder lists..... -> done :-)
FEC folders...... -> Backup is already there!
HMI folders...... -> done :-)
Shadow files..... -> Backup is already there!
lsd.jxe.......... -> done :-)
Dataset's........ -> Backup is already there!
EEPROM........... -> Backup is already there!
storage1......... -> Backup is already there!
storage2......... -> Backup is already there!
RCC Flashlock.... -> done :-)
RCC.............. -> done :-)
MMX.............. -> done :-)
IFS-stage2.......4512+0 records in
4512+0 records out
 -> done :-)
OFFSET's......... -> done :-)
backing up backup -> Backup part 1 is already there!
ls -als /net/mmx/mnt/boardbook --> total 192
      64 drwxrwxrwx  3 root      root          32768 Jan 01 00:02 .
      64 drwxrwxr-x  5 root      root          32768 Jan 01 00:02 ..
      64 drwxrwxrwx  3 root      root          32768 Jan 01 00:11 MU0809
Free space on /net/mmx/mnt/boardbook 884M

Patch folder /net/mmx/fs/sda0/patches/MHI2_US_POG11_K3300_MU0809_PATCH NOT found on SD card

-----------
timestamp: 2024_11_15_17_29_45

RCC-carp---->
CarPlay and Android Auto activated now

-----------
timestamp: 2024_11_15_17_29_50

RCC-addimage---->
File checksum does not match. No patch applied
Button images are already installed

-----------
timestamp: 1970_01_01_00_08_40

MMX-flash---->

MMX-flash---->

MMX-flash---->
ifs-root-part2 (ASCII): 00f0
3 image type=2(IFS     )  ba0000...1a8a1ff (15639040)|93...213 (120):LOCKED
0x00ba0000 - 00A2EE00

Found JTB-07408.04.16119301DF Backup, will check if all complete...
Standard-Backup will require ~5 minutes to complete!
Partition info... -> Backup is already there!
SWDL............. -> Backup is already there!
Version info..... -> Backup is already there!
GEM info........ -> Backup is already there!
Folder lists..... -> Backup is already there!
FEC folders...... -> Backup is already there!
HMI folders...... -> Backup is already there!
Shadow files..... -> Backup is already there!
lsd.jxe.......... -> Backup is already there!
Dataset's........ -> Backup is already there!
EEPROM........... -> Backup is already there!
storage1......... -> Backup is already there!
storage2......... -> Backup is already there!
RCC Flashlock.... -> Backup is already there!
RCC.............. -> Backup is already there!
MMX.............. -> Backup is already there!
IFS-stage2....... -> Backup is already there!
OFFSET's......... -> Backup is already there!
backing up backup -> Backup part 1 is already there! -> Backup part 2 is already there!

ls -als /net/mmx/mnt/boardbook --> total 192
      64 drwxrwxrwx  3 root      root          32768 Jan 01 00:02 .
      64 drwxrwxr-x  5 root      root          32768 Jan 01 00:02 ..
      64 drwxrwxrwx  3 root      root          32768 Jan 01 00:11 MU0809
Free space on /net/mmx/mnt/boardbook 884M

Patch folder /net/mmx/fs/sda0/patches/MHI2_US_POG11_K3300_MU0809_PATCH NOT found on SD card

beeso79, 2025-04-02:
Hi
I have exactly the same issue, I have patched the system to activate CarPlay, but when trying to connect the phone it does not recognize it as a USB device, whereas charging is available.
I am wondering how to fix this.

aleks123o, 2025-06-11:
I also have this problem, install went all good and I have the option to connect my phone for car play, but it doesn't recognize it when plugged in. IPod also doesn't show up anymore when it did before the update.

spg383, 2025-08-03:
Did any of you find a solution for this issue? I have the same problem.