Mr-MIBonk/M.I.B._More-Incredible-Bash #605: Missing Patch MU1326-MHI2_ER_AU37x_P5169

vinicius-trev, 2024-11-29:
Latest available patch found is for: MHI2_ER_AU37X_P5153

Any ideas if the P5169 is a > release than P5153? Should I need to "downgrade" to P5153?

Full backup can be found [here](https://drive.google.com/file/d/1RGVY9zwX4Rp5A2EZ5sZHBmsBMx_faS_6/view?usp=sharing).

mrvagbr, 2025-01-25:
Hello Vinicius, were you able to patch over version P5169 or did you need to downgrade to P5153?

vinicius-trev, 2025-01-25:
I didn't try the patch since the versions were different, should I try the P5153 patch on the P5169 or downgrade first?

jamesmccarthy, 2025-11-11:
> I didn't try the patch since the versions were different, should I try the P5153 patch on the P5169 or downgrade first?

@vinicius-trev did you try the previous version's patch files, or did you wait? Or was there another solution?

kiki54140, 2025-11-12:
> I didn't try, kept mine MMI at the OEM version since I can't confirm if it
> will work or not

Do you have the update? Thank you, because I'm getting error b200500.