Mr-MIBonk/M.I.B._More-Incredible-Bash #610: Mibwiki.one is down

guizmo84170, 2024-12-17:
Hello, i need download MTSD activator by enthusiast but wiki is down. 

Can you check it or send me thé files please ! 

Thx a lot

ScottYarosh, 2024-12-27:
I was just about to download everyting and also noticed all the links for the patch files are down.

haimiko, 2024-12-29:
Still down.  Any alternate sources?

2226cc, 2025-01-09:
Anyone know if this will be coming back?
I'd be willng to host if need be.

haimiko, 2025-01-09:
> Anyone know if this will be coming back? I'd be willng to host if need be.

Just keep trying.  Took me half dozen times to finally get through. Server seems extremely overloaded.

2226cc, 2025-01-09:
Asked on the telegram group if it's permanently down and answer to that was yes.

haimiko, 2025-01-09:
What file are you looking for?  I downloaded the vehicle patches last week if that's what you're looking for.

guizmo84170, 2025-01-09:
I search MTSD activator by enthusiast

JCviggen, 2025-04-28:
> What file are you looking for? I downloaded the vehicle patches last week if that's what you're looking for.

@haimiko  any chance you could get me all the euro porsche patches? if by email, you can take my username here and just add google's free email service domain to it..

Any help appreciated.

haimiko, 2025-05-10:
I only downloaded the MHI2_US_POG11_K5186_1_MU1476_AIO_20220723 file.  is
that what you need?