Mr-MIBonk/M.I.B._More-Incredible-Bash #681: Mibsolution site down (Need Patch file)

Dusk-RS7, 2025-04-27:
Unable to get last files for apple carplay, I just need "MHI2_US_AU57x_K3341_MU1426_Patch" the very last step needed.  If anyone can please give me there file or get the site back up. I would greatly appreciate it 

[image]

mmsa-tech, 2025-04-28:
Here are: 
https://drive.google.com/file/d/1kV5GaN52SeFBwNTK11n_VpSdxzbknZqs/view?usp=sharing

I don't know what's going on with the mibsolution website, but I hope it will be back up and running soon.

Dusk-RS7, 2025-04-29:
thank you, @mmsa-tech I will give it a try and see if it works. If I have any issues can I be able to message you

Maltman3000, 2025-05-02:
MMSA-tech can you also supply MHI2_ER_POG11_P3299?

zfonseca-1, 2025-05-02:
Also interested in MHI2_ER_POG11_K3299_MU0807_PATCH pls! :)

Maltman3000, 2025-05-02:
> Also interested in MHI2_ER_POG11_K3299_MU0807_PATCH pls! :)

https://www.upload.ee/files/18045511/MHI2_ER_POG11_P3299_MU0807_PATCH.zip.html

WelsonLi, 2025-05-02:
Hi @mmsa-tech may I ask if you also have "MHI2Q_US_AUG22_P5087_MU1316"

zfonseca-1, 2025-05-02:
> > Also interested in MHI2_ER_POG11_K3299_MU0807_PATCH pls! :)
> 
> https://www.upload.ee/files/18045511/MHI2_ER_POG11_P3299_MU0807_PATCH.zip.html

Thank you! 
Took a few clicks in the link to get the right file :)

man388, 2025-05-03:
Hello,
 I would be grateful if someone could help me with this :)
 I am looking for: MHI2_ER_AUG22_P3346 (MU: 1438) patch.

mmsa-tech, 2025-05-03:
> Hello, I would be grateful if someone could help me with this :) I am looking for: MHI2_ER_AUG22_P3346 (MU: 1438) patch.

_P3346 patch does not exist

mmsa-tech, 2025-05-03:
> Hi [@mmsa-tech](https://github.com/mmsa-tech) may I ask if you also have "MHI2Q_US_AUG22_P5087_MU1316"

i have only MHI2Q_ER_AUG22_P5092_MU1329

man388, 2025-05-03:
> > Hello, I would be grateful if someone could help me with this :) I am looking for: MHI2_ER_AUG22_P3346 (MU: 1438) patch.
> 
> _P3346 patch does not exist

My current version is: MHI2_ER_AUG22_S2116 MU0333

https://mib-helper.com say: 
2. Latest available firmware updates:
Latest official firmware update: MHI2_ER_AUG22_P3346 (MU: 1438, [details](https://mib-helper.com/index.php?train=MHI2_ER_AUG22_P3346)).
Latest possible firmware update: MHI2_ER_AUG22_P3346 (MU: 1438, [details](https://mib-helper.com/index.php?train=MHI2_ER_AUG22_P3346)).

I see information in other sites about using: MHI2_ER_AUG22_K3346 MU1438.

@mmsa-tech maby you know which patch would be good for me?

keykeyflex, 2025-05-03:
Hello @mmsa-tech , i have a VW Tiguan and i need the firmware MHI2_ER_VWG11_K3342, could you help me to download it please?
Thanks in advance

Amli1724, 2025-05-04:
Hi @mmsa-tech can you also supply

Trank you 

[image]

Maltman3000, 2025-05-04:
Full collection

https://mega.nz/file/GQg03AaQ#13wJKlvsfbuuFHQUxtHuhqlMShz-chDfNSotGMRzhq4

keykeyflex, 2025-05-04:
> Full collection
> 
> https://mega.nz/file/GQg03AaQ#13wJKlvsfbuuFHQUxtHuhqlMShz-chDfNSotGMRzhq4

Thank you so much, l have found the patch MHI2_ER_VWG11_K334_MU1427 in your pack.
My current firmware is MHI2_ER_VWG11_P3250.
To update m'y car's firmware, i need the firmware MHI2_ER_VWG11_K3342.
I guess the patch isn't enough to do the update, isn't it?
Thanks in advance.

Amli1724, 2025-05-04:
/

> Full collection
> 
> https://mega.nz/file/GQg03AaQ#13wJKlvsfbuuFHQUxtHuhqlMShz-chDfNSotGMRzhq4

Thank you! you are the best 👍🏽

zfonseca-1, 2025-05-05:
> Full collection
> 
> https://mega.nz/file/GQg03AaQ#13wJKlvsfbuuFHQUxtHuhqlMShz-chDfNSotGMRzhq4

Thanks a lot!

david-a-kelly, 2025-05-06:
Howdy guys, does anyone have access to fhe following?

MH2P_ER_AUG35_P9805

RamisA4q, 2025-05-08:
I am looking for latest firmware update : MHI2_ER_AUG22_P3346 (MU: 1438)

riuhasz, 2025-05-09:
Hi, I am looking for an older update, but since the mibsolution website is down, I thought of asking in here, maybe someone has it. The firmware I'm looking for is for is my Golf 7 (2013), 1st version MIB - **MSTD_EU_VW_P5301** - probably the last available update for this MIB ver. Thank you

PepTa, 2025-05-10:
>@mmsa-tech 
> i have only MHI2Q_ER_AUG22_P5092_MU1329

Can you/someone please share the Firmware-MHI2Q_ER_AUG22_P5092_MU1329 because the mibsolution.one is not working.

TY

insecret, 2025-05-15:
Hello,
Help me please, I searched everywhere but I can't find. Would someone be kind and share the firmware : MOI3_EU_VWMQB_P33000P   ?

JeniCzech92, 2025-05-19:
Hi guys, as the mibsolution.one is still down, does anybody happen to have datasets relevant for Dynaudio retrofit for Passat B8? Many thanks!

guizmo84170, 2025-05-19:
Hello i search MST2_EU_SE_ZR_P0253T please

witold1955, 2025-05-20:
Witam Wszystkich!
Skoro mibsolution.one wciąż nie działa, czy ja mógłbym prosić ktoś z Państwa o przesłanie danych istotnych do modernizacji MMI z Bang&Olufsenem w Audi Q5 z 2019r. Chodzi mi o plik MHI2Q_ER_AUG22_P5152_MU1329+PATCH. Pozdrawiam i wielkie dzięki!

SutoGabor97, 2025-05-23:
Anyone can send me one of this SW?
MST2_EU_VW_P0231D
MST2_EU_VW_P0883D

@Maltman3000  or du you have all of the MST2 firmware like earlier MHI2?
Thanks in advance!

KrzePikPL, 2025-05-24:
I'm looking for MST2_EU_SE_ZR_P0516T but google is silent as well as mibsolution

Hassiji, 2025-06-10:
> Here are: https://drive.google.com/file/d/1kV5GaN52SeFBwNTK11n_VpSdxzbknZqs/view?usp=sharing
> 
> I don't know what's going on with the mibsolution website, but I hope it will be back up and running soon.

> > Hello, I would be grateful if someone could help me with this :) I am looking for: MHI2_ER_AUG22_P3346 (MU: 1438) patch.
> 
> _P3346 patch does not exist

Hello, i'm looking for the firme : MST2_EU_SE_ZR_P0516T
Thanks a lot

JetGinger985, 2025-06-20:
Can someone please share the firmware MHI2_JP_POG11_P0627_MU0627_PATCH? 

mibsolution is not online yet.  Andthe Full collection includes MHI2_JP_POG11_P0625_MU0625_PATCH, but it looks like an older one.

venkyg88, 2025-06-20:
I need MU0726-MHI2_US_POG11_P3276 patch please

AceSour7, 2025-07-06:
[image]

The collection doesn't include:
MHI2_ US_POG11_K3300_MU0809_PATCH

Can someone please share it?
Thank you!

ricocarlotelli318, 2025-10-21:
Hallo ich suche MHI2_ER_VWG11_K3342 (MU: 1427, firmware update link für download.
 danke schonmal im vorraus

Radoslav-Molov, 2025-10-21:
> Hi, could somebody share MHI2_ER_AU37X_P5089 with me if they have it. I would be very happy.
> 
> Thanks in advance

[MHI2_ER_AU37x_P5089_MU1326_PATCH.zip](https://github.com/user-attachments/files/23031867/MHI2_ER_AU37x_P5089_MU1326_PATCH.zip)

raymond789521, 2025-10-22:
good morning people, the mibsolution website seems to be down still, my current train is MST2_EU_VW_P0867D and i need MST2_EU_VW_P0883D to get my s23u to work with the nav, the vw dealer cannot help me unfortunately so i have to do this myself. hoping someone here can help me to obtain the update files please.

Radoslav-Molov, 2025-10-24:
[MHI2_ER_AUG22_K3346_MU1438_PATCH.zip](https://github.com/user-attachments/files/23124795/MHI2_ER_AUG22_K3346_MU1438_PATCH.zip)

> > > Full collection
> > > https://mega.nz/file/GQg03AaQ#13wJKlvsfbuuFHQUxtHuhqlMShz-chDfNSotGMRzhq4
> > 
> > 
> > Thank you so much, l have found the patch MHI2_ER_VWG11_K3342_MU1427 in your pack. My current firmware is MHI2_ER_VWG11_P3250. To update m'y car's firmware, i need the firmware MHI2_ER_VWG11_K3342. I guess the patch isn't enough to do the update, isn't it? Thanks in advance.
> 
> Unfortunately, link no longer works... at least for me :/ could you by any chance share patches for MHI2_ER_AUG22_K3346?

Radoslav-Molov, 2025-10-24:
> Hello can anyone help me with MHI2Q_ER_AUG22_P4212 Patch ?
> 
> thanks

[MHI2Q_ER_AUG22_P4212_MU1177_PATCH.zip](https://github.com/user-attachments/files/23130171/MHI2Q_ER_AUG22_P4212_MU1177_PATCH.zip)

wdavoile-rgb, 2025-10-25:
Hi, does anyone have MST2_EU_VW_ZR_P0480T
Thanks

OliTV, 2025-10-25:
Hey!
Does anyone have **MEN2_EU_SKGPX_P9160L (MU: 8742)** saved?
I'm looking for this update because audio through Apple CarPlay randomly crackles, and when I use the radio (e.g. pushing _SETUP_ button) the crackling stops for a few seconds and then starts again. I think I'm gonna go crazy.
Thanks in advance!

Ash051201, 2025-10-25:
Hi,

Would like to find out if anyone has any AIO files for MHI2_ER_POG11_K5126_1_MU1394 AIO?

Thanks

Radoslav-Molov, 2025-10-25:
> Hi,
> 
> Would like to find out if anyone has any AIO files for MHI2_ER_POG11_K5126_1_MU1394 AIO?
> 
> Thanks

I have the patch only

Davew87, 2025-10-28:
Hi
Does anyone have the MIB patch file for mhi2q_er_aug22_p5092?
If so, that would be greatly appreciated as MIB.solution is down still

Radoslav-Molov, 2025-10-28:
> Hi Does anyone have the MIB patch file for mhi2q_er_aug22_p5092? If so, that would be greatly appreciated as MIB.solution is down still

[MHI2Q_ER_AUG22_P5092_MU1329_PATCH.zip](https://github.com/user-attachments/files/23193773/MHI2Q_ER_AUG22_P5092_MU1329_PATCH.zip)

mikajsalo, 2025-10-29:
Hi.
I would need MST2_EU_VW_PQ_p0356t.
Can anyone share?

Radoslav-Molov, 2025-10-29:
[MHI2Q_ER_AUG22_P5152_MU1329_PATCH.zip](https://github.com/user-attachments/files/23219669/MHI2Q_ER_AUG22_P5152_MU1329_PATCH.zip)

> > > Hi Does anyone have the MIB patch file for mhi2q_er_aug22_p5092? If so, that would be greatly appreciated as MIB.solution is down still
> > 
> > 
> > [MHI2Q_ER_AUG22_P5092_MU1329_PATCH.zip](https://github.com/user-attachments/files/23193773/MHI2Q_ER_AUG22_P5092_MU1329_PATCH.zip)
> 
> Do you have a MHI2Q_ER_AUG22_P5152 ??
> 
> Thank you!

SebAtkins, 2025-10-30:
Hi everyone,

Does anyone have the patch file for MHI2_ER_AU43X_P5098?

Thanks :)

FrankO1188, 2025-10-30:
Good afternoon all. To state the obvious, the MIB patch site is STILL down. Does anyone have the full patch set available? as the link above has expired.

I need to patch MHI2_US_POG24_K5136 1416 for widescreen carplay with Bose.

Thank you in advance for any assistance.

andrejhorak, 2025-10-31:
Hi guys, looking for a **MHI3_ER_AU_P4318** however I am not sure if I can update directly from **MHI3_ER_AU_P3393**.

Planned to do it in increments, however mibsolution is down for at least a week so I have no access to files :(

Thanks for help!

Zraulzz, 2025-11-01:
hi, somebody have MST2_EU_VW_P0231D i would like update i cant find this version

nikolica91, 2025-11-01:
> Cijela kolekcija
> 
> https://mega.nz/file/GQg03AaQ#13wJKlvsfbuuFHQUxtHuhqlMShz-chDfNSotGMRzhq4

If anyone has the full collection of patches, I’d really appreciate it if you could send it. Thanks in advance.

sxlencebyte, 2025-11-01:
Hello I search a files for MEN2_EU_SEGPX_P8010L
If somebody can help me pls :) ?

rushmoremods, 2025-11-04:
can someone help me get the aio MHI2_US_POG24_K5136 1416  for this AIO
i need to update the firmware first before installing the patch, please someone help, been trying for 2 weeks

jerryromero666, 2025-11-08:
MST2_US_VW_P0879D if possible :)

redoinezouitin-lang, 2025-11-18:
@keykeyflex Thanks a lot!!!!!

matthieugrollier-cloud, 2026-08-17:
Hi everyone,

i'm looking for the MEN2_EU_VWGPX_P8050L 

thanks a lot

bruno-monteiro5, 2026-10-16:
> 5\. > Also interested in MHI2_ER_POG11_K3299_MU0807_PATCH pls! :)
>     
>     
>     https://www.upload.ee/files/18045511/MHI2_ER_POG11_P3299_MU0807_PATCH.zip.html

Does someone have this patch still?

sat306, 2026-10-26:
Есть ли у когонибудь MH2P_ER_AUG35_P9805 ?

danpetvaldsky, 2026-11-04:
Guys, someone have **MEN2_EU_SKGPX_P9160L (MU: 8742)??** Thank you very much!! Need help :(

Donald-vld, 2026-11-05:
If mibsolution.one is down, u can use mibvault.com :)

danpetvaldsky, 2026-11-05:
I double checked and these version MEN2_EU_SKGPX_P9160L is no visible :/

manylittlestorms, 2026-11-14:
I am searching for MEN2_EU_VW_P4211L or MEN2_EU_VW_P5040L. Any help? I am desperate.