Mr-MIBonk/M.I.B._More-Incredible-Bash #682: Audi 2016 C7.5 Carplay not working after MIB and MMI firmware update

Dusk-RS7, 2025-04-28:
I had updated my MMI firmware on my 2016 Audi RS7 from "MHI2_US_AUG11_S0533" to this version "MHI2_US_AU57X_K3341_MU1426" to activate Carplay. Everything was updated including My SDARS and LTE portion all updated. Next I installed "MIB 3.6.0 QR Version" including patches and dragged into (patches folder) installed it again through engineering menu was successful. Lastly I went into G.E.M selected M.I.B and did AIO Patch for carplay. I now have the selection for (Audi Smartphone Interface) on my MMI but when I try to plug my phone in on either USB 1 or 2 no prompts show up that there is a device connected. If someone can help it would very very appreciative.  

[image]

[image]

[image]

[MU1426-JTB-07408.07.1511761031.log](https://github.com/user-attachments/files/19933845/MU1426-JTB-07408.07.1511761031.log)

jtlettko, 2025-06-30:
I am in the exact same boat and am looking for solutions. Hopefully someone has run into this before and can comment... i am going to keep digging, unless you've been able to find anything?

gojiito, 2026-10-02:
@Dusk-RS7 Hey! I found your issue because I'm doing the exact same S0533 → K3341/MU1426 upgrade on my 2016 A7. I'm stuck at the same point because I'm missing the MHI2_US_AU57x_K3341_MU1426_PATCH, specifically the 3098EE00 IFS file that your log shows.

Do you still have that patch or know where I can get it? I'd really appreciate it!