Mr-MIBonk/M.I.B._More-Incredible-Bash #686: MIB Solution offline

Hdudhfud, 2025-05-01:
Hi, I’m currently having trouble accessing mibsolution.one — the site seems to be down.

I’m trying to enable Apple CarPlay on my VW Tiguan (free activation), but I’m missing the update file for my model with the SW Train: MST2_EU_VW_ZR_P0472T, which contains Apple CarPlay.

I really need this specific update to proceed. If anyone has a working link or can share it, I’d greatly appreciate the help!

Thanks in advance!

[image]

Hdudhfud, 2025-05-01:
is there even an update file like in delphi

rockyy996, 2025-05-08:
Anybody knows what happened to mibsolution.one ?

Fanaticjonathanjordan, 2025-05-21:
There will be no more mib solution the servers are all down because many profited from mib solution